Azure/azure-sphere-samples #138: Temperature is modelled as double but is being reported as string.

valluriraj, 2020-05-28:
void SendSimulatedTelemetry(void)
{
    // generate a simulated temperature
    static float temperature = 50.0f;                    // starting temperature
    float delta = ((float)(rand() % 41)) / 20.0f - 1.0f; // between -1.0 and +1.0
    temperature += delta;

    char telemetryBuffer[TELEMETRY_BUFFER_SIZE];
    int len = snprintf(telemetryBuffer, TELEMETRY_BUFFER_SIZE, "{\"Temperature\":\"%3.2f\"}",
                       temperature);
    if (len < 0 || len >= TELEMETRY_BUFFER_SIZE) {
        Log_Debug("ERROR: Cannot write telemetry to buffer.\n");
        return;
    }
    SendTelemetry(telemetryBuffer);
}

johnnyhli, 2020-05-29:
Hi @valluriraj - Can you help me understand more about your question? Thanks.

valluriraj, 2020-05-29:
[image: image.png]

this is the code change.

Raj Valluri

valluriraj, 2020-06-02:
data being sent as part of temperature update is in string format. Datatype for temperature is declared as double in sphere template  (cloud side).  Because of the datatype mismatch certain functions (export data)  may not work properly. 
[image]

johnnyhli, 2020-08-13:
Fixed [here ](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/main.c#L856)and closed

closed 2020-08-13 by johnnyhli