Azure/azure-sphere-samples #139: Multiple Cores GPIO Conflict

mehrdadh, 2020-06-03:
I have two applications, one running on A7 and the other running on M4 core.
On A7, I request permission to all LED GPIOs and on M4 I request GPIO12 which is the **button A GPIO**. Both applications cannot coexist because there is an GPIO conflict. I receive this error:
**error: Could not deploy application to device: Application manifest requests a GPIO that is already in use by another application.**

I traced it and it seems this is the conflict:
**A7  --> MT3620_RDB_LED2_RED --> MT3620_GPIO15 --> 15
M4 --> buttonAGpio --> 12**

Is this a bug or expected?

My setup:
OS: Ubuntu 18.04
Azure Sphere CLI:  20.04
Azure Sphere OS version: 20.04 Update 2
API: 5Beta

johnnyhli, 2020-06-03:
Hi @mehrdadh - Thanks for bring this up. I'll look into it.

johnnyhli, 2020-06-03:
Hi @mehrdadh - I think this [link](https://docs.microsoft.com/en-us/azure-sphere/app-development/gpio-rt#mt3620-support-for-gpios-on-the-real-time-cores) may help you. 
>On the MT3620, most GPIOs are mapped to real-time cores in blocks of four, starting with GPIO 0; however, ISU blocks 0 - 4 have five GPIOs. As a result, RTApp GPIO requests may fail if a high-level application (or another RTApp) requests a GPIO in the same block. For example, if a high-level app requests GPIO 8 and an RTApp requests GPIO 9, the second app returns an error at deployment.

Please let me know if you have another questions.
 
Thanks,
Johnny

mehrdadh, 2020-06-04:
Hi @johnnyhli 
It makes sense since both 12 and 15 would be in the same block.
Thanks for clarification.

closed 2020-06-04 by mehrdadh