Azure/azure-sphere-samples #140: sphere sample application AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR

RotateAt60MPH, 2020-06-06:
I'm trying to run the sphere sample application in azure-sphere-samples\Samples\AzureIoT, but I keep getting the above error.  I'm running it in VSCode against a test application I created in IoT Central.

I was able to build and run the sample in azure-sphere-samples/Samples/HTTPS/HTTPS_Curl_Easy successfully.  

I was following the directions in **IoTCentral.md**, which included the page "**Set up Azure IoT Central to work with Azure Sphere**".

This is what I get when I run the sample application:

```
Starting debugger....

Process /mnt/apps/819255ff-8640-41fd-aea7-f85d34c491d5/bin/app created; pid = 99
Listening on port 2345

Remote debugging from host 192.168.35.1, port 57798

Azure IoT Application starting.

Using Azure IoT DPS Scope ID 0ne00116A38
Opening SAMPLE_BUTTON_1 as input.
Opening SAMPLE_LED as output.

[Azure IoT] Using HSM cert at /run/daa/787a2daa-0293-4009-8942-6be87b3eaa3a

IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR
'.
ERROR: Failed to create IoTHub Handle - will retry in 60 seconds.
```

I have the latest Sphere OS, and I do have network connectivity:

```
Your device is running Azure Sphere OS version 20.05.
The Azure Sphere Security Service is targeting this device with Azure Sphere OS version 20.05.
Your device has the expected version of the Azure Sphere OS: 20.05.

SSID                : ##########
Configuration state : enabled
Connection state    : connected
Security state      : psk
Frequency           : 5220
Mode                : station
Key management      : WPA2-PSK
WPA State           : COMPLETED
IP Address          : 10.168.1.63
MAC Address         : 2c:f7:f1:08:a0:e1
```

Here's my app_manifest.json:

```
{
  "SchemaVersion": 1,
  "Name": "AzureIoT",
  "ComponentId": "819255ff-8640-41fd-aea7-f85d34c491d5",
  "EntryPoint": "/bin/app",
  "CmdArgs": [
    "0ne001<#####>"
  ],
  "Capabilities": {
    "AllowedConnections": [
      "global.azure-devices-provisioning.net",
      "iotc-cf1ed8fe-2907-43d2-823f-37ec21d710c9.azure-devices.net",
      "iotc-0aa42664-e375-4e53-98f3-4eebffa00fb7.azure-devices.net",
      "iotc-b69366dc-dd8e-4bb0-abbd-0462ee1381bc.azure-devices.net",
      "iotc-2388c46f-a656-42f1-8e0e-144587e8c88b.azure-devices.net",
      "iotc-7a74f745-0964-4a14-a2f4-85680dd0beeb.azure-devices.net",
      "iotc-3d4254f0-bc08-470a-890e-e4e680d6b4d2.azure-devices.net",
      "iotc-e43113b6-6c87-4984-a180-5bb24e075a4c.azure-devices.net",
      "iotc-d9b75e2f-1072-40dd-9ae0-e7777063e70e.azure-devices.net",
      "iotc-41c84b39-0ca8-4545-a21d-b9ada42dab58.azure-devices.net",
      "iotc-0f5e03c5-fad2-40ce-8385-7b704760f0e9.azure-devices.net"
    ],
    "Gpio": [
      "$SAMPLE_BUTTON_1",
      "$SAMPLE_LED"
    ],
    "DeviceAuthentication": "787a2daa-0293-4009-8942-<############>"
  },
  "ApplicationType": "Default"
}
```

Any suggestions appreciated.

RotateAt60MPH, 2020-06-06:
Found the problem.  There was a step missing in "Run the sample with Azure IoT Central" (IoTCentral.md) page.  In section: 'Configure the Azure IoT Central application', you create a Device Template, but you should also create a Device!  That appears to be the issue.  I'll leave this open for now and someone can decide what to do with it.

johnnyhli, 2020-06-08:
Hi @RotateAt60MPH  - Thanks for asking the question. When I run the sample, I didn't manually create device. After building and running the app, you should find a new device under **Device / All devices**, and then follow [Show your device data in Azure IoT Central](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTCentral.md#show-your-device-data-in-azure-iot-central) to migrate your device to your template. 

I think [Troubleshooting](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTCentral.md#troubleshooting) would help you.

>'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.'
>This error may occur if:
>The setup for Azure IoT Central or Azure IoT Hub has not been completed

Please let me know if the problem still there.

Thanks,
Johnny

closed 2020-07-06 by johnnyhli