Azure/azure-sphere-samples #141: Networking_IsNetworkingReady saying network is not ready

lokkitshmike, 2020-06-08:
I was just wondering, that Networking_IsNetworkingReady() does state, that network is not ready, while the AzSphere CLI says that everything is fine:

`azsphere dev wifi show-status

SSID                : pmySSID
Configuration state : enabled
Connection state    : connected
Security state      : psk
Frequency           : 5200
Mode                : station
Key management      : WPA2-PSK
WPA State           : COMPLETED
IP Address          : 192.168.1.90
MAC Address         : 2c:f7:f1:08:73:de`

I'm using the code from the AzureIoT sample:
```
    bool isNetworkReady = false;
    if (Networking_IsNetworkingReady(&isNetworkReady) != -1) {
        Log_Debug("Networking is not ready...\n");
        }
```

johnnyhli, 2020-06-09:
Hi @lokkitshmike - I think this [link](https://docs.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-networking/function-networking-isnetworkingready) can help you.

>Verifies whether internet connectivity is available, the device is connected to an access point, has an IP address, and the time is synced.
> It does not check whether there is actual internet connectivity.

floresboy, 2020-06-09:
same issue here...

lokkitshmike, 2020-06-09:
@johnnyhli thanks for replying. I've read the documentation. From my point of view it isn't very useful that the function only returns bool. information It should return something that indicates the exact problem. **But aside this, the main problem is**: What should I do if I would know the exact problem?

- is connected to an access point: Obviously **it is**, because AZSphere dev wifi show-status says so
- has an IP address: Obviously **it has**, because AZSphere dev wifi show-status says so
- and the time is synced: Don't know, but how should I check this and what would I do if not?
- It does not check whether there is actual internet connectivity: It seems to **have** connectivity, because with the AZSphere CLI, I can do all tasks that require connectivity (e.g. device recovery)

Another weird fact: I tried the sample ~a week ago and it worked perfectly. Yesterday I just re-ran the same code and it stopped working because networking says to be not ready.

As mentioned I also tried to reset the device with `AZSphere dev recover` - still no luck.

floresboy, 2020-06-09:
Issue is present in examples: https://github.com/Azure/azure-sphere-samples/issues/105

lokkitshmike, 2020-06-09:
@floresboy  Oh wow! After reading that and because I excluded any other issues in my case, it really seems to break down to an RTC/timesync issue.

Without knowing the current time, the MCU is not able to check whether a SSL certificate is valid or not. I think this is the reason, why HTTP is working while HTTPS is not.

But anyways, I would have expected, that the MCU is pulling the time from any Microsoft server upon boot.
I'll investigate further into that to see which system time the MCU has and if it tries to ntp-sync at boot.

floresboy, 2020-06-09:
Good analysis I think but in the example if I remove the SSL part (of HTTP only) AND remove the connection check the example works. So the check is faulty even without the time I guess.. Thanks for your efforts

Kamiel

johnnyhli, 2020-06-09:
Thanks @lokkitshmike  and @floresboy for your comments.
@lokkitshmike , do you mind to let me know more information so that I can better help you?   

1. run "azsphere get-support-data -o data.zip" and send this output and the zip file to [AzCommunity@microsoft.com](mailto:AzCommunity@microsoft.com) (do not attach it here), with a link back to this issue
1. If it is OK, please let me know the brand and model of your WiFi router
1. If it is OK, please let me know your general location (e.g. North America, East Europe, etc.)

Thank you so much!

lokkitshmike, 2020-06-09:
@johnnyhli I've created the support data archive and sent it to AzCommunity@microsoft.com.

I am working with this router: https://avm.de/produkte/fritzbox/fritzbox-6590-cable/ and I am connected via a wired repeater: https://avm.de/service/fritzwlan/fritzwlan-repeater-1750e/uebersicht/
As you might expect by the router brand, I am based in Germany.

I did some more research and wrote a little debug routine. **The problem definitely breaks down to a system time issue**. When I am powering the board (no RTC battery attached) the clock starts at 0 and the systime is not going to be synced, even though the boars HAS connectivity and TimeSync Service states to be enabled.

This is my debug routine:
```
const struct timespec sleepTime = { .tv_sec = 1, .tv_nsec = 0 };
void* foo(void* arg) {

    while (true) {

        time_t rawtime;
        struct tm* timeinfo;
        time(&rawtime);
        timeinfo = localtime(&rawtime);
        Log_Debug("Current local system time and date: %s", asctime(timeinfo));

        bool isTimeSyncEnabled = false;
        Log_Debug("Calling Networking_TimeSync_GetEnabled()...");
        if (Networking_TimeSync_GetEnabled(&isTimeSyncEnabled) == -1) {
            Log_Debug("call FAILED\n");
        }
        else {
            Log_Debug("call SUCCEEDED\n");
        };
    
        Log_Debug("Checking return value of Calling Networking_TimeSync_GetEnabled()...");
        if (isTimeSyncEnabled) {
            Log_Debug("TimeSync IS enabled!\n");
        }
        else {
            Log_Debug("TimeSync is NOT enabled...trying to enable TimeSync...");
            if (Networking_TimeSync_SetEnabled(true)) {
                Log_Debug("SUCESS!\n");
            }
            else {
                Log_Debug("FAILED!\n");
            }
        }

        Log_Debug("Calling Networking_IsNetworkingReady()...");
        if (Networking_IsNetworkingReady(&isNetworkingReady) == -1) {
            Log_Debug("call FAILED: %s (%d)\n", strerror(errno), errno);
        }
        else {
            Log_Debug("call SUCCEEDED.\n");
            Log_Debug("Checking return value of Networking_IsNetworkingReady()...");
            if (isNetworkingReady) {
                Log_Debug("networking IS ready.\n");
            }
            else {
                Log_Debug("networking is NOT ready.\n");
            }
        }
        nanosleep(&sleepTime, NULL);
    }
}
```

And it repeatedly produces this output:

```
Current local system time and date: Sat Jan  1 00:00:12 2000
Calling Networking_TimeSync_GetEnabled()...call SUCCEEDED
Checking return value of Calling Networking_TimeSync_GetEnabled()...TimeSync IS enabled!
Calling Networking_IsNetworkingReady()...call SUCCEEDED.
Checking return value of Networking_IsNetworkingReady()...networking is NOT ready.
```
**BUT, I just tried to connect the board to my iPhones hotspot and that worked perfectly and is reproducable**:

- delete all wifi profiles from the board
- power down the board
- power the board
- run the debug routine: as expected: systime is near 0 and neworking is not ready
- add the iPhone hotspot as wifi profile (debug is still giving wrong time an networking not ready)
- wait a little until the board associates to the wifi
- debug routine shows the correct time and that networking is now ready:

```
Current local system time and date: Sat Jan  1 00:01:27 2000
Calling Networking_TimeSync_GetEnabled()...call SUCCEEDED
Checking return value of Calling Networking_TimeSync_GetEnabled()...TimeSync IS enabled!
Calling Networking_IsNetworkingReady()...call SUCCEEDED.
Checking return value of Networking_IsNetworkingReady()...networking is NOT ready.
Current local system time and date: Tue Jun  9 20:13:44 2020
Calling Networking_TimeSync_GetEnabled()...call SUCCEEDED
Checking return value of Calling Networking_TimeSync_GetEnabled()...TimeSync IS enabled!
Calling Networking_IsNetworkingReady()...call SUCCEEDED.
Checking return value of Networking_IsNetworkingReady()...networking IS ready.
```

When I do this with adding my "normal" wifi instead of the iPhones hotspot, time is NOT going to be synced! That is also reproducable. The weird part about it is, that it worked perfectly when I connected the board to my "normal wifi" some days ago.

**Is there some more documentation on how the OS is trying to timesync?**

floresboy, 2020-06-09:
so connection over the iPhone data connection is resulting in proper time setting while the Fritzbox (good equipment btw) is not working; mysterious

johnnyhli, 2020-06-09:
Hi @lokkitshmike - Thanks for help us narrow down the problem. We are currently looking into this issue. 

The device is unable to sync time with the Microsoft NTP time server (prod.time.sphere.azure.net, time.sphere.azure.net).  There could be a few reasons:

1.	Traffic on UDP ports 123 (outgoing) or 124 (incoming) is blocked or being dropped (see https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains).  Could you capture how your FritzBox’s NTP options are set and pass them back to us?
2.	DNS resolution is failing.  The above names should resolve to time.windows.com which in turn usually resolves to a hostname somewhere on .akadns.net.
3.	The NTP reply message is unexpected and Sphere rejects it

It will also help us if you could capture a network trace of traffic to/from the Azure Sphere device.
> Windows: https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-server-2012-R2-and-2012/jj129382(v=ws.11)
> Linux:  Wireshark, etc.

lokkitshmike, 2020-06-10:
@johnnyhli I tried another dev (different model) board on another network (Cisco Meraki wifi) and it worked. 
But anyway I am a little confused about what the board does after startup (MAC SichuanA_25:31:81 is the dev board)

- sending out 19x the same ARP request (probing):

```
10.989500	SichuanA_25:31:81	Broadcast	ARP	60	Who has 6.143.87.230? (ARP Probe)
.
11.128486	SichuanA_25:31:81	Broadcast	ARP	60	Who has 6.143.87.230? (ARP Probe)
```
BTW 6.143.87.230 belongs to "Headquarters, USAISC (HEADQU-3)" which seems to be "United States Army Intelligence and Security Command" - so what is this???

- sending out 19x the same DHCP discover packets
```
12.674033	0.0.0.0	255.255.255.255	DHCP	312	DHCP Discover - Transaction ID 0xab92aa0f
.
12.813368	0.0.0.0	255.255.255.255	DHCP	312	DHCP Discover - Transaction ID 0xab92aa0f
```

After it gets an IP and gateway from the DHCP server it also sends out 19x ARP requests for the gateway IP.
After that, the board seems to act normal. It then does the NTP sync, msftconnecttest and some encyrypted stuff.

Why is the board doing this? from my point of view it's not normal behaviour to do so. It should only send one ARP/DHCP request and wait for a response!?
I could imagine that my FritzBox rates this behaviour as flooding the network and therefore is locking the board down. I will try to figure that out...

pietrobr, 2020-06-23:
I'm starting having the same issues, never happen to me until now

johnnyhli, 2020-06-24:
Hi @lokkitshmike - Did you get chance to run "azsphere get-support-data -o data.zip" again for SichuanA_25:31:81 board at your office?   And do you mind to send the zip file to AzCommunity@microsoft.com with a link back to this issue. That would help us better working with this issues.

johnnyhli, 2020-06-24:
Hi @pietrobr - Thanks for bring this up. To help us better understand the issue you have, can you let us know more about what sample apps or scenario you are  running? Also, if it is OK, can you:

1. run "azsphere get-support-data -o data.zip" and send this output and the zip file to AzCommunity@microsoft.com (do not attach it here), with a link back to this issue
1. Please let us what dev board you are using
1. Please let us know the brand and model of your WiFi router
1. Please let us know your general location (e.g. North America, East Europe, etc.)

Thank you so much!

Best,
Johnny

lokkitshmike, 2020-06-29:
@johnnyhli Any news on this? Still not able to get it to work in my normal Wifi...

pietrobr, 2020-06-29:
I am not able to upload at the moment, but I have no problem if I run the samples using my iPhone, but if I have my home WiFi if doesn’t work. Just point out that I have never had that problem in the past with my home wifi

johnnyhli, 2020-06-29:
Hi @lokkitshmike, we are still working on this issue. But I can't reproduce one of your issue: I have no issue running ```w32tm /stripchart /computer:ptbtime1.ptb.de``` and ```
w32tm /stripchart /computer:prod.time.sphere.azure.net``` Do you have issue running these command under which router? I saw you mentioned that board works on Cisco Meraki wifi.  And does "normal WiFi" mean Cisco Meraki  WiFi?  Thanks.

johnnyhli, 2020-07-06:
Hi @pietrobr - Do you mind share what hardware you are working on and device OS version? Also can you try if the issue still occurs while running the latest OS? Thank you so much!

pietrobr, 2020-07-07:
Hi @johnnyhli good news, I've updated my MT3620 Dev Kit to 20.06 OS and now I don't  have the issue mentioned before!

johnnyhli, 2020-07-13:
Hi @lokkitshmike - Can you also try  if the issue still occurs while running the 20.06 OS? Thank you so much!

johnnyhli, 2020-08-13:
Hi @lokkitshmike - We appreciate any feedback, but we haven't had any response from you. Therefore, we are going to close this issue. If you still have issues, please feel free to reopen this issue. Thank you.

closed 2020-08-13 by johnnyhli