Azure/azure-sphere-samples #142: FPU on Cortex-M4

mehrdadh, 2020-06-09:
I wasn't sure if this is the right place to ask question about Cortex-M4 core, so let me know if I should post this issue somewhere else.
I'm trying to build a simple program on M4 that uses **softfp** or **hard** floating point unit. Here is my example:

```
_Noreturn void RTCoreMain(void)
{
App_Init();
UART_Printf(debug, "test init\r\n");
float test = 1.0;
float test1 = test + 2.0;
}
```
And here are my flags:
```
SET(C_FLAGS     "-Wall -Wextra -ffunction-sections -fdata-sections -fno-delete-null-pointer-checks -Wconversion -fsingle-precision-constant -MMD -O0 -g1 -march=armv7e-m -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp")
SET(CXX_FLAGS   "-Wall -Wextra -ffunction-sections -fdata-sections -fno-delete-null-pointer-checks -Wconversion -fsingle-precision-constant -MMD -O0 -g1 -march=armv7e-m -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp")
SET(ASM_FLAGS   "-x assembler-with-cpp -Wall -Wextra -ffunction-sections -fdata-sections -fno-delete-null-pointer-checks -Wconversion -fsingle-precision-constant -MMD -O0 -g1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp")
SET(LD_FLAGS    "-Wl,--verbose -Wl,--start-group -Wl,--wrap,main -Wl,--gc-sections -Wl,--end-group -march=armv7e-m -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp")
```
This will result in a hardfault exception on the summation. I believe the issue is with **-mfloat-abi=softfp**. Any help is appreciated here.
Thanks

johnnyhli, 2020-06-09:
Hi @mehrdadh - Thanks for bring this up. We will look into it.

aiden-ct, 2020-07-07:
Any updates on this?

matsujirushi, 2020-07-07:
Is it helpful?

https://github.com/MediaTek-Labs/mt3620_m4_software/blob/master/MT3620_M4_Sample_Code/FreeRTOS/MT3620_RTApp_FreeRTOS_CPP/AzureSphereRTCoreToolchainMTK.cmake#L52-L53

johnnyhli, 2020-07-10:
Hi @mehrdadh and @aiden-ct - Sorry for the long wait. The floating point unit on M4 are turned off by default. Here are the steps to turn on this feature:

1. Add this to the RT app code to turn on the floating point support on the M4 (as per these [ARM instructions](https://developer.arm.com/documentation/100166/0001/floating-point-unit/fpu-programmers--model/enabling-the-fpu)):
```
// Enable the FPU.
// ARM 100166_0001_00_en S7.3.2
__asm("ldr.w r0, =0xE000ED88"); // CPACR
__asm("ldr r1, [r0]");
__asm("orr r1, r1, 0xF << 20"); // [20:23] enable CP10 and CP11
__asm("str r1, [r0]");  
```        

2. Add this to the RT app's CMakeLists.txt to get the compiler/linker to use this support:
```
target_compile_options(${PROJECT_NAME} PUBLIC -mfloat-abi=hard -mfpu=fpv4-sp-d16)
target_link_libraries(${PROJECT_NAME} PUBLIC -lm -mfloat-abi=hard -mfpu=fpv4-sp-d16)
```

Please let me know if you have other questions.

Thanks,
Johnny

mehrdadh, 2020-07-12:
@johnnyhli Thanks for the solution. I'll try this next week and let you know.

johnnyhli, 2020-07-22:
Hi @mehrdadh - Do you still have other questions? If not, I will closed this issue. Thanks.

closed 2020-07-23 by mehrdadh