Azure/azure-sphere-samples #143: Improve document by inserting a picture

xiongyu0523, 2020-06-15:
In this [section](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTCentral.md#configure-the-sample-application-to-work-with-your-azure-sphere-tenant-and-devices), could you add a picture to show what a correct app_manifest.json will be? Now our bootcamp material is referring to this page for default Azure sphere template setup, many attendees do not know how to set it correctly. And there is no word to mention that user need fill scopeID to CmdArgs key. 

I used to use this picture, or something similar
[image]

johnnyhli, 2020-06-15:
Hi @xiongyu0523 - Thanks for bring this up! We will look into it. 

I think the ShowIoTCentralConfig program could also help you with this issue. If you followed [Configure the sample application...](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTCentral.md#configure-the-sample-application-to-work-with-your-azure-sphere-tenant-and-devices) , the ShowIoTCentralConfig program will generate a configuration code like this:

```
Find and modify the CmdArgs, AllowedConnections and DeviceAuthentication lines in your app_manifest.json so each includes the content from the below:
"CmdArgs": [ "0ne0xxxxxx" ],
"Capabilities": {
    "AllowedConnections": [ "global.azure-devices-provisioning.net", "iotc-xxxx.azure-devices.net" ],
    "DeviceAuthentication": "--- YOUR AZURE SPHERE TENANT ID---",
}

Obtain your Azure Sphere Tenant ID by opening an Azure Sphere Developer Command Prompt/Terminal and typing the following command:

'azsphere tenant show-selected'
```

Will this configuration code help? 

Thanks,
Johnny

xiongyu0523, 2020-06-19:
Better we can highlight what fields customer need modify . This is the most error prone area.

closed 2020-07-06 by johnnyhli