Azure/azure-sphere-samples #146: Regarding guideline in private network services example

kiranpradeep, 2020-07-08:
This is not an issue ticket, but regarding a guideline made in private network services sample. In 3rd paragraph of README.md [[1](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/PrivateNetworkServices/README.md#sample-private-network-services)], the guideline is to authenticate and encrypt the connection for production use case. But to encrypt the connection a TLS library should be part of SDK which is not the case. The developer cannot encrypt the connection with out TLS library in the SDK. Or is there TLS library already part of SDK? Kindly point to TLS library if it exists.

johnnyhli, 2020-07-13:
Hi @kiranpradeep  - Thanks for bring this up. We understand the request and we hope to offer a TLS API soon.   

Thanks,
Johnny

johnnyhli, 2020-07-30:
Hi @kiranpradeep - Azure Sphere version 20.07 is now available.  Azure Sphere applications can now use the wolfSSL client-side TLS API. For more information please check [here](https://techcommunity.microsoft.com/t5/internet-of-things/azure-sphere-version-20-07-is-now-available/ba-p/1542749). Thanks.

closed 2020-07-30 by kiranpradeep