Azure/azure-sphere-samples #147: Choice of target name in CMake

WeiDaiWD, 2020-07-08:
Function `azsphere_target_add_image_package` is defined in "AzureSphere.cmake" and calls function `_azsphere_target_add_image_package_common` defined in "AzureSphereInternal.cmake". Both functions requires a `target` argument. But it seems that `_azsphere_target_add_image_package_common` uses both `${target}` and `${PROJECT_NAME}` to refer to the CMake target. Both values evaluate to the same thing if `azsphere_target_add_image_package(${PROJECT_NAME})` is used. It throws an error if `azsphere_target_add_image_package(something_else)` is used.

The following two points are not clearly stated in documentation or error messages:
1. The CMake target passed to `azsphere_target_add_image_package` must be named `${PROJECT_NAME}`.
2. There must be only one invocation of `azsphere_target_add_image_package`.

johnnyhli, 2020-07-10:
Hi @WeiDaiWD - Thanks for bring this up. We are currently working on this.

johnnyhli, 2020-07-30:
Hi @WeiDaiWD - Azure Sphere version 20.07 is now available. We updated the [documentation].(https://docs.microsoft.com/en-us/azure-sphere/app-development/using-cmake-functions#image-package-creation)
>The CMake target passed to azsphere_target_add_image_package must be named ${PROJECT_NAME}, and the azsphere_target_add_image_package function can be called only once from the CMakeLists.txt file.

Please let me know if you have other questions.

closed 2020-07-30 by johnnyhli