Azure/azure-sphere-samples #149: Loading project in VS2019 gives error

TAMHAN, 2020-07-13:
Hello Folks,
both the Azure Sphere sample available from the VS2019 generator and the sample "azure-sphere-samples-master\Samples\HelloWorld\HelloWorld_HighLevelApp" found in this repository cannot be loaded on my German-language workstation running Version 16.7.0 Preview 3.1 of VS2019.

If I use the CMake Loader, I always get the error string "Der Index war außerhalb des Arraybereichs", and the project does not load. Instead, the system only shows the CMakeList contents in an editor window.

johnnyhli, 2020-07-14:
Hi @TAMHAN - Thanks for bring this up. We will look into it.

johnnyhli, 2020-07-14:
Hi @TAMHAN  - Could you try install non-preview version VS2019 and check if you still have the same issue? In [Visual Studio 2019 Preview Release Notes](https://docs.microsoft.com/en-gb/visualstudio/releases/2019/release-notes-preview), it states that:
>This release is not "go-live" and not intended for use on production computers or for creating production code. For instructions on installing and updating Visual Studio 2019, see this documentation on [updating Visual Studio 2019 to the most recent release](https://docs.microsoft.com/en-us/visualstudio/install/update-visual-studio?view=vs-2019).

Thanks,
Johnny

TAMHAN, 2020-07-15:
Hello,
good news. My workstation greeted me with an Update to Preview 4 (Version 16.7.0 Preview 4.0) today. All works fine now. Please close this issue.

closed 2020-07-15 by TAMHAN