Azure/azure-sphere-samples #151: [Question] Material for Ethernet on M4

Visit-Learn, 2020-07-21:
Hi,

I know that Azure RTOS is very popular on IoT and currently we want to evaluate it how could we implement
internet ability on MT3620 M4 (such as using ENC28J60 chip). We know that Azure RTOS has very good implementation
on internet in Azure RTOS NetX, so we want to know there has any material to describe MT3620 M4 internet implementation
on Azure RTOS?

Thank you so much.

johnnyhli, 2020-07-22:
Hi @Visit-Learn - Thanks for asking the question. Here is the sample code for [Azure Sphere with Azure RTOS](https://github.com/Azure-Samples/Azure-RTOS-on-Azure-Sphere-Mediatek-MT3620). 

>This sample demonstrates how Azure Sphere and Azure RTOS are able to run together on MT3620 Development Kit.

Hope this is help.

Thanks,
Johnny

johnnyhli, 2020-07-28:
Hi @Visit-Learn - Regarding the question of using Azure RTOS for internet connectivity via the MT3620 M4, we highly recommended that all your internet connectivity is via the Azure Sphere OS  on the high-level core.  This will ensure you benefit from all of the security benefits that Azure Sphere provides and that distinguish it from other IoT solutions.  Further, internet connectivity must be via the high-level core in order for the device to receive OS/app updates, to acquire its certificate, and to send error reports. 

Please let me know if you have further questions. Thanks.

Visit-Learn, 2020-08-28:
Hi,

Thank you. But I have an idea about it, because as far as I know Azure RTOS ThreadX is versatile with MCU level internet ability (NetX). And in IoT world, there may have low level sensor with simple TCP stack which we could use NetX abliity with M4 to communicate with those sensors and trigger mailbox to A7 then sending data with security benefits that Azure Sphere.

And I think that have some benefits for above architecture,
1. Real time
2. Dedicated (M4 dedicated on sensor  data capture)
3. HLApp (A7 for real world Internet communication)
4. Power consumption (In the future, I think that MT3620 could wake up M4 only to capture sensor data)

What's about your thinking, thanks a lot.

johnnyhli, 2020-09-03:
Hi @Visit-Learn - [Guardian modules](https://docs.microsoft.com/en-us/azure-sphere/hardware/guardian-modules) could be an alternative solution.  The [Private Ethernet](https://docs.microsoft.com/en-us/azure-sphere/network/use-network-services) support in the Azure Sphere OS allows a connection to the brownfield device over Ethernet without exposing it to the public internet. Placing the device in the [Power Down state](https://docs.microsoft.com/en-us/azure-sphere/app-development/power-down) when it is inactive can also drastically reduce power consumption and thus increase battery life.

Visit-Learn, 2020-09-09:
Hi,

Thank you for your information, I will close this case in a few days if there has no other questions.
Thanks a lot.

closed 2020-09-17 by Visit-Learn