Azure/azure-sphere-samples #154: Why all samples use event loops which is comparatively hard to follow?

kiranpradeep, 2020-07-24:
Almost all high level app samples in the repository are based on epoll and event-loops. Now comparing AzureIoT telemetry samples in Azure-Sphere [[1](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/main.c)] vs normal azure-iot-sdk-c [[2](https://github.com/Azure/azure-iot-sdk-c/blob/master/iothub_client/samples/iothub_ll_telemetry_sample/iothub_ll_telemetry_sample.c)] repository, the event-loop based approach is not as easy to read and follow. Why this repository is not providing programs in conventional linear approach as in azure-iot-sdk-c? Similar there is no sample using pthreads either. Is that a guideline for Sphere developers to follow?

johnnyhli, 2020-08-20:
Hi @kiranpradeep - Sorry for the long wait. We are reviewing your question and expect to get back to you soon. Thanks.

johnnyhli, 2020-08-27:
Hi @kiranpradeep - Here is the answer for your question:

Yes, Azure Sphere event loops are our recommended pattern for a few reasons:
	
1. Azure Sphere event loops are required with Azure Sphere for managing system events like OS and application update notifications. You must use event loops to [defer device updates](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fazure-sphere%2Fapp-development%2Fdevice-update-deferral&data=02%7C01%7Creturok%40microsoft.com%7C1e3444d24f164948b5f108d8490b8f86%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637339661285617647&sdata=hQzdv%2FScZ4Z%2FWuS%2Bk9toxOBRx2VdXUKueAr7BPxTvAA%3D&reserved=0). 
2. Event loops allow for asynchronous programming (multiple tasks can progress concurrently). 
3. Event loops minimize memory overhead. Every thread carries the memory overhead of a separate stack. Linux stacks are allocated with virtual memory and grow on-demand to an upper limit so there is no fixed cost. Threads are therefore harder to implement on a constrained system like the Azure Sphere MT3620 because of the additional resource cost. 
4.Synchronization between threads is complex and can lead to issues like deadlocks. Event loops are simpler in this regard. 

The linked IoT C SDK sample is a simple sample written to be platform agnostic and showcase an API. Event loops are platform specific and a pattern we want to showcase with Azure Sphere as the best way to develop Azure Sphere applications.

Yes, linear samples with threads would be easier for the comprehension of an Azure Sphere API. However, linear code doesn’t scale well as an app grows in complexity under the constrains of the platform. Basically the samples would be easier to understand but less practical to reuse or build upon for real-world apps. So the samples are serving this double-purpose of showing how an API works but also how to write scalable code for the platform.

It's also worth noting that the Azure Sphere samples use our [EventLoop API](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fazure-sphere%2Freference%2Fapplibs-reference%2Fapplibs-eventloop%2Feventloop-overview&data=02%7C01%7Creturok%40microsoft.com%7C1e3444d24f164948b5f108d8490b8f86%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637339661285627646&sdata=tAugO%2FgkYcfk22aCBbzjdUAHTjG6OVqhb9nRrbjky%2Bk%3D&reserved=0) which is a bit different from event loops in general.

1. Event loops are a model of waiting on file descriptors and are a standard UNIX pattern. This is not an Azure Sphere specific model - it's a common pattern in UNIX/Linux code. Early on, we decided to use a programming pattern based on epoll (which we chose over select and poll because it has lower overhead).
 
2. The [EventLoop API](https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.microsoft.com%2Fazure-sphere%2Freference%2Fapplibs-reference%2Fapplibs-eventloop%2Feventloop-overview&data=02%7C01%7Creturok%40microsoft.com%7C1e3444d24f164948b5f108d8490b8f86%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637339661285637636&sdata=lEnpIvYUpP79YFXenI7OM8%2FwLJp7F%2BqdSDR2uCEAvIE%3D&reserved=0) is a specific implementation of event loops which is required in Azure Sphere applications to receive our system notifications like events related to update.

kiranpradeep, 2020-08-28:
Thanks for helpful answer. It was really help if you can answer one last question: Why not remove the pthread [API](https://docs.microsoft.com/en-us/azure-sphere/app-development/threads-concurrency) altogether? What is its intended usecase? 

I will raise future queries at https://docs.microsoft.com/en-us/answers/topics/azure-sphere.html. This question was raised only only since that is a continuation query.

johnnyhli, 2020-08-28:
Hi @kiranpradeep - Thanks for asking the question. We will review your question and get back to you soon.

johnnyhli, 2020-09-18:
Hi @kiranpradeep - Sorry for the late reply.

Here is the answer for your question:

Developers might want to use pthreads for situations where parallel/asynchronous execution is desired, and it is not possible to do this in a non-blocking way (as required/recommended when using EventLoop). For example, if you have some data processing code that takes some time to run and need to keep servicing a responsive UI at the same time. 

Separately, to address your comment on why we include pthreads while also providing EventLoops, there are certain functionalities that are useful in certain scenarios. We want to provide developers the flexibility to leverage APIs of their choice while providing guidance on our recommendations (EventLoops).

closed 2020-09-18 by kiranpradeep