Azure/azure-sphere-samples #155: sphere tenant certificate chain has varying key sizes

kiranpradeep, 2020-07-27:
During testing mutual authentication, my https server was not able to validate the azure sphere client certificate. I tried validate the leaf certificate (received in https server application) with tenant certificate chain (`azsphere tenant download-ca-certificate-chain --output CA-cert-chain.p7b`) with openssl and it failed. 
```
openssl verify -verbose -CAfile CA-cert-chain.cer leaf.cer 
C = US, ST = Washington, L = Redmond, O = Microsoft Corporation, CN = Microsoft Azure Sphere 9b831aeb-bbee-408f-b2dc-210e4538c5ad
error 7 at 1 depth lookup: certificate signature failure
error leaf.cer: verification failed
```

It appeared that tenant ca cert had key size of 256 while sphere root ca and policy ca had key size of 384. Is that ok? Attaching tenant certificate chain from avnet mt3620 device running 20.06.

Attaching leaf and tenant chain as [certs.zip](https://github.com/Azure/azure-sphere-samples/files/4983652/certs.zip).

johnnyhli, 2020-07-28:
Hi @kiranpradeep - Thanks for asking the question. We are currently looking into it.

johnnyhli, 2020-07-29:
Hi @kiranpradeep - [20.07 SDK](https://techcommunity.microsoft.com/t5/internet-of-things/azure-sphere-version-20-07-is-now-available/ba-p/1542749)  is now available. Could you download the latest SDK from [here](https://docs.microsoft.com/en-us/azure-sphere/install/install-sdk?pivots=visual-studio#install-the-azure-sphere-sdk) and try the new [azsphere ca-certificate](https://docs.microsoft.com/en-us/azure-sphere/reference/azsphere-ca-certificate) command? Thanks.

kiranpradeep, 2020-07-30:
No. That did not help. Kindly find below screenshot and [certs](https://github.com/Azure/azure-sphere-samples/files/4998632/certs.zip) used.
[image]

johnnyhli, 2020-07-30:
Hi @kiranpradeep - the root cause is the cert is malformed and need to be renewed. [Here ](https://techcommunity.microsoft.com/t5/internet-of-things/known-issue-azure-sphere-tenant-ca-certificate-rotation/ba-p/1557169)is more detail about this issue. The issue will be automatically fixed starting 5th August. But, if you are looking to renew the cert right away, we are happy to help you with it. If you want us to renew the cert now, please sent your Device ID and Tenant ID to AzCommunity@microsoft.com (do not attach it here), with a link back to this issue. Thanks.

kiranpradeep, 2020-07-31:
Usually when reporting a valid issue to Github projects, the owner of the repository, gives a minimal response in the lines of "Thanks for reporting this issue. I have fixed it commit #xxxxxxx". But here (and in the linked techcommunity article written today) the response was as if, the issue was independently found and fixed. Had that been the case, I would not have been asked to check again after 20.07 release. Not a concern. Just saying :)

johnnyhli, 2020-08-03:
Hi @kiranpradeep - Thank you for reporting this issue, it really helped us find and fix this issue. With your thorough information, the team was able to target the issue fast. Secondly, I do apologize for bringing more confusion at the beginning here by suggesting you upgrade your SDK.  I misunderstood the root cause for your issue and provided the wrong solution.  Thank you so much for your understanding. Best, Johnny

kiranpradeep, 2020-08-11:
@johnnyhli I tried today 11/August. The tenant certificate chain validation still fails. Am I doing anything unexpected? 

[image]

iamjohnnyli, 2020-08-11:
Hi @kiranpradeep -  Sorry for the inconveniences. From the [article](https://techcommunity.microsoft.com/t5/internet-of-things/known-issue-azure-sphere-tenant-ca-certificate-rotation/ba-p/1557169) I shared, the new certificate will be activated after Aug 18th.

>For tenants that are impacted by this issue, the new and valid tenant CA certificates will be created by August 5, 2020. The new certificates will be activated after August 18, 2020. 

Again, we are happy to help. If you want us to renew the cert now, you can sent your Device ID and Tenant ID  to [AzCommunity@microsoft.com](mailto:AzCommunity@microsoft.com)(do not attach it here), with a link back to this issue.

Thanks,
Johnny

kiranpradeep, 2020-08-12:
@johnnyhli Thanks for the reply. I was confused for your reply above on 31/Aug.

> Hi @kiranpradeep - the root cause is the cert is malformed and need to be renewed. Here is more detail about this issue. The issue will be automatically fixed starting 5th August. But, if you are [...]

closed 2020-08-12 by kiranpradeep