Azure/azure-sphere-samples #156: Network is not ready

piercalderan, 2020-07-28:
Even if the wifi status shows the connection is OK, the AzureIoT sample rarely connects to Azure IoT Central...
I try to claim device, show-status, start and restart the app but most of the times it shows the message:
WARNING: Network is not ready. Device cannot connect until network is ready.
Please help.

SSID                : Alice
Configuration state : enabled
Connection state    : connected
Security state      : psk
Frequency           : 2412
Mode                : station
Key management      : WPA2-PSK
WPA State           : COMPLETED
IP Address          : 192.168.1.183
MAC Address         : 2c:f7:f1:08:81:16

johnnyhli, 2020-07-28:
Hi @piercalderan  - Thanks for bring up the question. In order to better help you, could you provide more information? Do you mind share what hardware you are working on, what WIFI router you are using, and which version of device OS? Thanks.

piercalderan, 2020-07-29:
Hi. Thanks for reply. I partially solved the problem using my hotspot from my smartphone. It's a very strange story, because in this way it works most of the times. Using the home wi-fi, it doesn't. Aniway, I think the problem is correlated to the wi-fi connection and not to azure sphere. I can close the issue.

closed 2020-07-29 by piercalderan