Azure/azure-sphere-samples #160: Azure IoT Sample Not working when not in debug mode

gearsmotion789, 2020-08-08:
I found that when I run the Azure IoT sample with a direct connection to the hub, it works just fine when I'm running in debugging mode through VSCode. However, when I reset the board or upload the program without debugging, it no longer works. How can I fix this?

bawilless, 2020-08-10:
I just confirmed this behavior using Visual Studio 2019.  When I use Visual Studio to run the Azure IoT example my application runs and connects to my IoT Hub without any issues.  If I don't run the example from Visual Studio it does not start/run.  

Previously to the 20.07 release, when debugging an application on my device, if I power cycle the device, my debug application would start and run automatically.  That's not happening either.

I just setup an OTA deployment for the Azure IoT example and verified that my application is installed onto my device (azsphere dev image list-installed). 
 The application is on my device, but it's not running.

I just did another test using the high level hello world example . . . 

With this example I see the expected behavior . . . 

- The application runs automatically in debug mode when I power cycle my device
- I added this image to my OTA deployment and confirmed that the application is pushed down OTA and runs

Something is broken either in the 20.07 release or the Azure IoT Example.  

Any ideas how to troubleshoot/fix this issue?

Thanks

cawhitworth, 2020-08-10:
Hi; thanks for raising this.

I've dug into it and it turns out to be an issue in the way the `CmdArgs` parameters from `app_manifest.json` are passed to applications on startup. Under the debugger, a `CmdArgs` parameter such as `--ScopeID 0ne0000000` is passed as two strings in argv, `--ScopeID` and `0ne0000000`. However, when the application is started normally, it is passed as a single string, `--ScopeID 0ne0000000`, which the argument parsing code then fails to parse.

For now, a workaround is to split all CmdArgs parameters with spaces in into multiple parameters. We will work on getting a fix into the code and docs shortly.

So, for example, where previously you might have had:
       `"CmdArgs": [ "--ConnectionType DPS", "--ScopeID <scopeIdOfYourDPS>" ]`

Change it to:
       `"CmdArgs": [ "--ConnectionType", "DPS", "--ScopeID", "<scopeIdOfYourDPS>" ]`

Thanks.

bawilless, 2020-08-10:
Awesome!  Thanks for the quick turn.

Brian

Brian

gearsmotion789, 2020-08-10:
Thanks. It works now.