Azure/azure-sphere-samples #163: [Quetsion] About lan-enc28j60-is board configuration resource usage

Visit-Learn, 2020-08-28:
Hi,

Currently I meet the issue, when I develop HL App with  lan-enc28j60-is board configuration is work fine.
After that, I move my stage into RT App and when I use GPIO7 in RT App, the device said that
**Device would not sideload app: Application manifest requests a GPIO that is already in use by another application (DeviceClientException)**

I have no idea why  lan-enc28j60-is board configuration uses GPIO7 and any document to describe it?
Thank you so much.

bawilless, 2020-08-28:
The MT3620 groups GPIOs into blocks.  If an application uses any one GPIO from a block, then that application has access to the other GPIOs in that same block, but other applications can't access any of the GPIO from the block that's being used, even if the first application is not using them.  I've seen a really nice explanation on this on-line somewhere, but I can't find it right now.  If I find it, I'll update this response.

Here's some more detail to make it clear . . . 

This table is from the MediaTek documentation: https://support.mediatek.com/AzureSphere/mt3620/M4_API_Reference_Manual/group__os__hal__gpioif__enum.html#ggaded1e1bd5b6050a69f7325f53e9577b4a2316d4bfeea19317cea0e0750f049fdf

The blocks are defined below:

GPIOIF group0(gpio0~3)
GPIOIF group1(gpio4~7)
GPIOIF group2(gpio8~11)
GPIOIF group3(gpio12~15)
GPIOIF group4(gpio16~19)
GPIOIF group5(gpio20~23)

Note that the lan-en28j60 image is using GPIO 5, from GPIOIF group1, so your application can't use GPIO7 in that same block. 

I hope this helps,

Brian

bawilless, 2020-08-28:
I just re-read your original message.  Your original high level application could use GPIO7, but when you moved it to the M4 it could not.  So I revise my original statement in regards to applications, the GPIO block restriction is across CPU cores, not applications, not applications.  I'll update my original answer.

johnnyhli, 2020-09-02:
Thanks @Visit-Learn for asking the question and thanks @bawilless for answering the question! I'll close this issue. If you still have the question, please feel free to reopen it. Thanks!

Visit-Learn, 2020-11-04:
> The MT3620 groups GPIOs into blocks. If an application on one CPU core uses any single GPIO from a block, then any application running on the same CPU core has access to the other GPIOs in that same block, but applications on other CPU cores can't access any of the GPIO from the block that's being used, even if applications on the first CPU core is not using them. I've seen a really nice explanation on this on-line somewhere, but I can't find it right now. If I find it, I'll update this response.
> 
> Here's some more detail to make it clear . . .
> 
> This table is from the MediaTek documentation: https://support.mediatek.com/AzureSphere/mt3620/M4_API_Reference_Manual/group__os__hal__gpioif__enum.html#ggaded1e1bd5b6050a69f7325f53e9577b4a2316d4bfeea19317cea0e0750f049fdf
> 
> The blocks are defined below:
> 
> GPIOIF group0: gpio0 - 3
> GPIOIF group1: gpio4 -7
> GPIOIF group2: gpio8 -11
> GPIOIF group3: gpio12 -15
> GPIOIF group4: gpio16 -19
> GPIOIF group5: gpio20 -23
> 
> Note that the lan-en28j60 image is using GPIO 5, from GPIOIF group1, this is running on the A7 CPU core. So your application running on the M4 CPU core can't use GPIO7 since it's in that same block.
> 
> I hope this helps,
> 
> Brian

Hi,

Could we know that GPIO24 - GPIO93 have same limitation?

Thanks a lot.

johnnyhli, 2020-11-04:
Hi @Visit-Learn, thanks for the question. I'll get back to you later.

johnnyhli, 2020-11-10:
Hi @Visit-Learn, here is the answer for your question: 

The same idea should be applied.
All the peripherals are grouped (like GPIO71~75 are in ISU4, GPIO41~48 are in ADC), and in core mapping, the interrupt/reset/dma are routed/mapped based on a group.
However, some pins, aside from those shared peripherals like GPIO85~88, GPIO89~92, GPIO49~52 are special since they’re dedicated for certain cores, so core mapping is not required, they’re enabled by default. However, as those pins serve for debugging, logging, or strapping.

Please let me know if you have other questions. Thanks!

closed 2020-11-10 by johnnyhli

Visit-Learn, 2020-11-10:
Hi,

Got it, thank you for your information~
Thanks a lot.