Azure/azure-sphere-samples #165: DeferredUpdate Sample Instructions are outdated

bawilless, 2020-09-17:
The instructions on the https://github.com/Azure/azure-sphere-samples/tree/master/Samples/DeferredUpdate/DeferredUpdate_HighLevelApp page need to be updated.  

Under Prepare the device to receive updates: 
* Step 4 is invalid, the device-group create command no longer accepts the --filepath argument
* Step 4 in invalid, creating a new "Field Test" device group in invalid since it's one of the default device groups created when the product was created
* Step 4 can be corrected by replacing the existing command with the command below
azsphere device-group deployment create -productname MyProduct -devicegroupname "Field Test" -filepath <path to blink imagepackage>

Under "Run the sample outside an IDE"
* After step 3 my device had two different user applications on the device, maybe because I never put the device back into cloud test mode?  I think the deferred update worked, I saw different LEDs light up on my Avnet Starter Kit.  It looked like the defer update application loaded, not the blink application.  But the blink application was OTA loaded to my device, it just was not automatically started.

C:\>azsphere dev image list-installed
Installed images:
 --> gdbserver
   --> Image type:   Application
   --> Component ID: 8548b129-b16f-4f84-8dbe-d2c847862e78
   --> Image ID:     3dcad309-0c08-43e5-b1e8-ffd2a758d061
 --> DeferredUpdate
   --> Image type:   Application
   --> Component ID: 541fc56d-1da1-452b-b0d3-dcac70700f59
   --> Image ID:     4670eea2-250b-4a69-97c5-742117e02e52
 --> HelloWorld_High
   --> Image type:   Application
   --> Component ID: 1689d8b2-c835-2e27-27ad-e894d6d15fa9
   --> Image ID:     ffd42f0b-cf4c-4978-8ace-bfe48688e673

I really wanted to see this example perform correctly, so I came up with these steps to make the demonstration bullet proof.  The user does not have to rely on good timing to see the update being deferred.

1. Create the new product
2. Update the deployment with the deferred update application
2.1. >azsphere dg deployment create -productname MyProduct -devicegroupname "Field Test" -filepath <path to deferred update imagepackage>
3. Move the device into the device group and put it into cloud test mode
3.1. azsphere device enable-cloud-test --productname MyProduct --devicegroupname "Field Test"
4. Verify that the device is connected to a wifi network
4.1 azsphere device wifi show-status
5. Power cycle the device to get the deferred update application onto the device
5.1. Verify that the deferred update application is loaded onto the device
5.2. azsphere dev image list-installed
6. Application runs and Green LED is lighted (deferring updates)
7. Update the deployment with the blink application
7.1. >azsphere dg deployment create -productname MyProduct -devicegroupname "Field Test" -filepath <path to blink imagepackage>
8. The application is not automatically notified that a new application is in the device group, it must "check-in" with AS3 to get this notification and that will happen on reboot or every 24 hours
9. Power cycle the device to force a check-in with AS3
10. The device restarts with the deferred update application running
11. The OS contacts AS3 and sees that there is an update pending
12. The application gets the notification and lights the SAMPLE_PENDING_UPDATE_LED LED (Blue LED on the Seeed board, User APP LED on the Starter Kits(after the change I'll submit today))
13. The user can press the A button to get the update
14. User presses the A button
14.1 The RGB led turns Yellow (Green + Red)
14.2 The deferred update is allowed to proceed
15. The device restarts and runs the blink application

This sequence does not allow the user to see the application debug, consider piping that debug out a UART so users that want to see it can connect the UART to a FTDI friend device.

johnnyhli, 2020-09-18:
Hi @bawilless - Thank you for bringing up this issue and providing the suggestion. We will investigate it.

johnnyhli, 2020-11-13:
Hi @bawilless - Thanks for the long waitting. We have updated the [README.md](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/DeferredUpdate/DeferredUpdate_HighLevelApp) file in  https://github.com/Azure/azure-sphere-samples/tree/master/Samples/DeferredUpdate/DeferredUpdate_HighLevelApp
Thanks again for bring this issue to our attention.

closed 2020-11-13 by johnnyhli