Azure/azure-sphere-samples #167: PrivateNetworkServices Client error

vaishnavils97, 2020-10-06:
When I tried the Azure sphere sample PrivateNetworkServices code,  I was getting a client error. I am using Azure sphere as the server and raspberrypi 3 as a client. If anyone could provide a sample client program it would be helpful for me to figure it out.

johnnyhli, 2020-10-06:
Hi @vaishnavils97 - Thanks for asking the question. What kind of client your were running and what error you had? The Private Network Serices Sample app configures the Azure Sphere device to run a DHCP server and an SNTP server, and implements a basic TCP server. You can run command to test all these servers. For example, to test the device's DHCP server, you can run `ip addr show`, `hostname --ip-address` or other command depends on what OS you are running on your Raspberry Pi. Thanks.

vaishnavils97, 2020-10-07:
I am running a buster OS on my Raspberry Pi device.  I actually want to use my Raspberry pi as a client and Azure Sphere as server .This scenario uses Socket to connect between client to server. So when the client starts to connect the server  getting an connect error(-1). TCP client is also running in Raspberry Pi, So if you could provide a sample client program I can figure out whether I am missing any aspects of communication.

johnnyhli, 2020-10-13:
Hi @vaishnavils97 - Sorry for the late reply. Have you tried the unmodified PrivateNetworkServices Sample App by following the [Test the application's TCP server](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/PrivateNetworkServices#test-the-applications-tcp-server)? If the sample works as expected, may I know what changes you made on the PrivateNetworkServices Sample App? Thanks!

johnnyhli, 2020-11-13:
Closed due to non-active for 1 month.

closed 2020-11-13 by johnnyhli