Azure/azure-sphere-samples #168: DeviceTwins not working for AzureIoT sample

snakechia, 2020-10-07:
I have follow the instruction to setup my Azure Sphere MT3620 by Seeed Studio.
Also setup my Azure IoT Hub accordingly. Everything work by design, until I want to "play" with the DeviceTwins function.

By changing the value for "**StatusLED**" from "**false**" to "**true**" it didn't light up the LED.
I did further investigation to the problem, noticed that the value of "**statusLedValue**' getting from Line 682 in the sample code it return "**-1**", it look like the DeviceTwins connection to Azure IoT Hub was not establish.

Have any ideas where goes wrong? Or I missed any step?
P/S: I'd running the same code on the same nearly a year ago on the same device. It was working before.

johnnyhli, 2020-10-07:
Hi @snakechia - Thanks for bring this up. I'll investigate it. Could you tell me what version of device OS and SDK you were using?  Thank you!

snakechia, 2020-10-07:
AzSphere device OS version: 20.07
SDK: Azure_Sphere_WinSDK_20_07

Dev PC:
OS: Windows 10 Build 19042.546
Visual Studio Enterprise 2019 (16.7.5)

johnnyhli, 2020-10-07:
Hi @snakechia - I just go through the IoT Hub DeviceTwins function. I use Device OS 20.09 and I was able to light up the LED. Do you mind update or recover the device to latest OS and try again?  Also make sure you add "StatusLED": true under "desired". Thanks!

snakechia, 2020-10-07:
Hi John,
Mysteriously, I re-clone the repo into a new local folder, forced update azsphere again, and run the code from a clean location, it now it work perfectly. 😅

Thanks.

johnnyhli, 2020-10-07:
Hi @snakechia - I'm glad it worked! I'll closed the issue. Please feel free to reopen it if you have further questions.

closed 2020-10-07 by johnnyhli