Azure/azure-sphere-samples #171: Testing Private Network Services sample not working

chuddles, 2020-11-11:
Right now I am trying to test the Private Network Services sample, specifically the TCP server. I have built and deployed the sample to my device (I am using a Microsoft Tahuya which uses Sphere hardware version seeed_mt3620_mdb), but when I open command prompt and attempt to connect to the Sphere's TCP server with "telnet 192.168.100.10 11000", I receive the error "Connect failed."

Some background: I commented out StartSntpServer and ConfigureAndStartDhcpSever in main.c because I wanted to set a static IP address and I don't need DHCP/SNTP for what I'm working on. My computer's IP address is statically set in the same domain.

Things I've tried: restarting and using Putty for Telnet instead of command prompt.

Any advice is appreciated. I am new to using the Sphere and TCP, so I may be missing something obvious here.

johnnyhli, 2020-11-12:
Hi @chuddles - Thanks for asking the question. I just tried commented out StartSntpServer and ConfigureAndStartDhcpSever in main.c in Private Network Services sample app. The hardware I use are Seed MT 3620 and [MT3620 Ethernet Shield](https://github.com/Azure/azure-sphere-samples/blob/master/HardwareDefinitions/mt3620_rdb/EthernetWiring.md). I can get the expect behavior. Have you tried the un-modified Private Network Services sample app? Also, can you try it again with the latest [20.10 SDK and OS](https://techcommunity.microsoft.com/t5/internet-of-things/azure-sphere-version-20-10-is-now-available/ba-p/1847018)? Thanks!

chuddles, 2020-11-12:
@johnnyhli thanks for your response. I updated the SDK and tried the un-modified Private Network Services sample (except for changing the hardware definition) and I am still getting the same issue (Telnet connect failed). Any ideas on additional things to try?

johnnyhli, 2020-11-12:
Hi @chuddles - do you mind to let me know more information, so that I can better help you?
If you don't mind, please run "azsphere get-support-data -o data.zip" and send this output, the zip file, and your modified hardware definition file to AzCommunity@microsoft.com (do not attach it here), **with a link back to this issue**.  Thanks!

chuddles, 2020-11-18:
Update: I've solved the problem. Here's what happened: 
1. Ethernet board configuration image package was not properly installed (I'm using a device that has an Ethernet plug built into the hardware so I incorrectly assumed that this image package was somehow already accounted for). I ran this command to fix it: azsphere image-package pack-board-config --preset lan-enc28j60-isu0-int5 --output enc28j60-isu0-int5.imagepackage
2. I switched to a different static IP address and subnet mask (for a different project) and then changed back to 192.168.100.X and 255.255.255.0, and after I did this it started working. I don't know what's up with that because they are the same IP address and subnet mask I was using before, but maybe this switch caused something to be reset. 

Thanks for the help @johnnyhli!

closed 2020-11-18 by chuddles

johnnyhli, 2020-11-18:
@chuddles - I'm so glad you resolved the issue. We will still look into the why changing the IP and subnet makes different. Thanks again for rise this up.