Azure/azure-sphere-samples #173: [wolfssl library] wolfSSL_CTX_set_verify() don't support

Shaing, 2020-11-13:
Hi,

Why don't support this function "wolfSSL_CTX_set_verify" of wolfssl library?

Thx,
A.H

johnnyhli, 2020-11-13:
Hi @Shaing - Thanks for asking the question. I'll look it up and get back to you.

johnnyhli, 2020-11-16:
Hi @Shaing, 

Microsoft has partnered with wolfSSL to provide a stable subset of features exposed through the Azure Sphere OS. By using these features, you get the benefit of wolfSSL without having to consider the memory impact of building it into your application (because it already ships with the OS) and the license (because Microsoft has already licensed it). A long term stability promise is important to Azure Sphere, and so we only expose some of the API items that wolfSSL has committed to for long term support (see wolfSSL's commitments here: https://www.wolfssl.com/wolfssl-and-application-binary-compatibility/). This way, an application that you build now will continue to run despite the OS receiving regular updates for years to come (including new versions of wolfSSL as they come out with security fixes). 
If you are willing to consider the memory impact and the licensing, you can always build wolfSSL into your application (see their guidance here: https://www.wolfssl.com/wolfssl-visual-studio-projects-microsoft-azure-sphere-mt3620-iot-mcu/) and receive a broader set of features. Unfortunately, these have to be mutually exclusive, you can either use the OS exposed API or bring you own, but cannot mix and match in your application. 
While we don't want to comment on future plans at this time, we will take into consideration that you value this feature for future support. 

Thanks again for the question! I'll close this issue. Please feel free to open it again if you have other questions.

closed 2020-11-16 by johnnyhli

josephalloyd, 2021-11-11:
Hi @Shaing! I wanted to follow up on this item even though its been a while. With our Azure Sphere 21.10 OS that is currently available, wolfSSL_CTX_set_verify() is now an OS exposed API!