Azure/azure-sphere-samples #174: Can you print prov_device_error and iothub_client_error we get certain failures from IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning

xiongyu0523, 2020-11-17:
Have seen customers reporting AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR, may be it is better to change from our sample level

Within SetUpAzureIoTHubClientWithDps()

```
    if (provResult.result == AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR) {
        Log_Debug("prov_device_error is %d.\n", provResult.prov_device_error);
    } else if (provResult.result == AZURE_SPHERE_PROV_RESULT_IOTHUB_CLIENT_ERROR) {
        Log_Debug("iothub_client_error is %d.\n", provResult.iothub_client_error);
    }
```

Or appreciate if you can add helper function to convert to string (like GetAzureSphereProvisioningResultString().

johnnyhli, 2020-11-18:
Hi @xiongyu0523 - Thanks for rise this up. We will look into it.

johnnyhli, 2020-11-18:
Hi @xiongyu0523  - just want to clarify what you asked for. Are you asking that the sample app should let SetUpAzureIoTHubClientWithDps() return string instead of true, false result? If not, can you explain more? Thanks!

xiongyu0523, 2020-11-20:
Yes, it should print the error log. Now the sample code only print one of the three result.

cawhitworth, 2021-07-05:
We no longer use the convenience API which returns AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR, and use the Azure IoT SDK-provided macros to decode errors returned by the SDK.

closed 2021-07-05 by cawhitworth