Azure/azure-sphere-samples #176: 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.

djaus2, 2020-11-23:
Remote debugging from host 192.168.35.1, port 16039
Azure IoT Application starting.
ConnectionType: DPS
ScopeID: 0ne00186D14
Using DPS Connection: Azure IoT DPS Scope ID 0ne00186D14
Opening SAMPLE_BUTTON_1 as input.
Opening SAMPLE_LED as output.
[Azure IoT] Using HSM cert at /run/daa/bfbe8861-c64c-4eae-a3f2-9adf312d2efd
IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR'.
ERROR: Failed to create IoTHub Handle - will retry in 60 seconds.

djaus2, 2020-11-23:
Seems similar to [#67](https://github.com/Azure/azure-sphere-samples/issues/67) but the solution there is not my problem as I didn't do that.

I get a similar error when doing same with IOTCentral with the app

djaus2, 2020-11-23:
Note that my tenant login is with a personal login, whereas the Azure login is a separate MSA account.  Could that be a problem?

djaus2, 2020-11-23:
Nb: Non AzureIoT apps run OK.

bawilless, 2020-11-23:
There's a blog that walks the user through all the Azure configuration steps [here](https://www.element14.com/community/groups/azuresphere/blog/2019/05/01/azure-sphere-starter-kit-out-of-box-demo-blog-post-2).  Note that the application documented in the blog does not use the same CmdArgs as you need for the AzureIoT Example, but all the details for the Azure Side are well documented.  

Here's a checklist of things to verify when using the AzureIoT application.  If any of these things are missing, you could see the response you're seeing.

1. IoT Hub setup
2. DPS setup
2.1 IoT Hub linked to DPS
2.2 Azure Sphere Tenant CA certificate uploaded to DPS
2.2.1 Tenant CA validated in DPS
2.3 Enrollment group setup in DPS
2.3.1 Enrollment group using validated Azure Sphere Tenant CA as primary certificate
3. AzureIoT application configuration
3.1 app_manifest.json entries updated to reflect your Azure Resources and your Azure Sphere Tenant
3.2 IoT Hub hostname listed in under "allowedConnections" list
3.2 Global DPS hostname "global.azure-devices-provisioning.net" "listed in "allowedConnections" list
3.3 Connection type and DPS Scope ID included in the "CmdArgs" entry 
"CmdArgs": [ "--ConnectionType", "DPS", "--ScopeID", "<your scope ID" ],
3.4 Your Azure Sphere Tenant GUID listed in the app_manifest.json "DeviceAuithentiction" field
3.4.1 azsphere tenant show-selected
4. The Azure Sphere device you are using is claimed in the same tenant as identified by the DeviceAuthentication field AND the Azure Sphere Tenant CA certificate uploaded/validated in DPS
5. The ports listed [here](https://docs.microsoft.com/en-us/azure-sphere/network/ports-protocols-domains) are open and routable on your wifi network.  Try using a different network to see it your device connects, like the hotspot on your mobile device.

Your Azure Sphere Tenant and Azure subsubscription do not need to be linked together in any way.  As long as you have access to manage each of these resources the example will work great.

johnnyhli, 2020-11-23:
Hi @djaus2 - Thanks for asking the question, and thanks for @bawilless for help. 

I suggest you don't share your Scope ID and other sensitive information here. I anonymized the information in your post. 

In [Troubleshooting](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTHub.md#troubleshooting), you can find that 'AZURE_SPHERE_PROV_RESULT_PROV_DEVICE_ERROR' may occur if [Azure  IoT Hub setup](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub) has not been completed. So, could you please double check again with the setup steps? Please pay attention to the links in the documentation. 

> Note that my tenant login is with a personal login, whereas the Azure login is a separate MSA account. Could that be a problem?

You can claim a device with Tenant A and then create an IoT hub with a completely different account, as long as you upload the right certs to the DPS/Hub.

djaus2, 2020-11-24:
Thanks for both of your comments. Thanks Johnny for my ID oversight.
Brian, That's a good list of gotcha's. 
Its good there is now a copy of that here.
I'll add it to my blog as well.

I am working through this but I suspect I might have a problem with
```2.3 Enrollment group setup in DPS```
Checking that out. Will update back here when done.

djaus2, 2020-11-24:
Worked through that again.
When using the  Hub my enrollment group wasn't set up properly. OK now.
Now to resolve for IOTCentral.

johnnyhli, 2020-11-24:
Hi @djaus2 - Glad it works for you! I'll close this issue. If you have any question related, please feel free to open it again.

closed 2020-11-24 by johnnyhli