Azure/azure-sphere-samples #178: Need to update app_manifest.json in AzureIoT Sample

djaus2, 2020-11-23:
_Change:_
"CmdArgs": [ "<IDScope"> ],
_To:_
"CmdArgs": [ "--ConnectionType", "DPS", "--ScopeID", "<IDScope" ],

johnnyhli, 2020-11-23:
Hi @djaus2 - Thanks for opening this issue. Can you help me understand more about your needs? The current default value in [app_mainfest.json](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/app_manifest.json) is empty and requires filling with different options depends on what scenario, e.g., IoT Central or IoT Hub, with DPS or without DPS.

djaus2, 2020-11-24:
**If I have in _ app_manifest.json_:**

 "CmdArgs": [ ],

**_upon compilation I get:_**

Severity	Code	Description	Project	File	Line	Suppression State
Error		The app_manifest.json 'CmdArgs' needs to contain the ConnectionType, Azure IoT Hub hostname and Device ID for Direct Connection, or ConnectionType and Scope ID for DPS connection.	

**_If I replace that with:_**

CmdArgs": [ "--ConnectionType", "DPS", "--ScopeID", "<IDScope" ]

**it compiles OK!**

**This applies to whether or not I am using IoT Central or IoT Hub.
I am not aware that using the app in both cases without DPS is an option.**

djaus2, 2020-11-24:
Nb I am using 20.10

djaus2, 2020-11-24:
Ok so these options ARE covered in the specific  ReadMes  IOTHub,IoTCentral, IoTEdge
Can I suggest then that the line:

```"CmdArgs": [ ]```
Be replaced with something like
```"CmdArgs": [ <See the ReadMe.>]```

I say this because what is now required has changed from when I first ran this sample about 6 months ago.

Or even use:
```"CmdArgs_IoTHub": [ "--ConnectionType", "DPS", "--ScopeID", "<IDScope>" ],
"CmdArgs_IoTCentral": [ "--ConnectionType", "DPS", "--ScopeID", "<IDScope>" ],
"CmdArgs_IoTEdge":```
With a note to select one.

johnnyhli, 2020-11-24:
Hi @djaus2, Thanks for the suggestion! We will look it up.

johnnyhli, 2021-01-14:
Hi @djaus2, Thanks again for your feedback. We will have a recent release to fix the issue. I'll let you know once it got fixed.

tokenstar, 2021-07-05:
@djaus2  thanks for the feedback.  We've decided not to put a placeholder in the app_manifest, but we have continued refining the readmes for this sample to help guide the user depending on which Azure IoT configuration is being targeted.  Please keep the feedback coming, it is welcome.

closed 2021-07-05 by tokenstar