Azure/azure-sphere-samples #181: IOTCentral with AzureIOT Sample: Getting the x509 certificate

djaus2, 2020-11-25:
Following the IOTCentral Readme

On [https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-central#step-2-download-the-tenant-authentication-ca-certificate](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-central#step-2-download-the-tenant-authentication-ca-certificate)

**Step 3. Upload the tenant CA certificate to Azure IoT Central and generate a verification code

1. OK
2. OK
3. Add and verify your primary X.509 certificate using the steps in the section to Add and verify a root or intermediate X.509 certificate.
4.

Q. Is there a simple way to create the x.509 certificate that can be used in 3. above. The instructions pointed to by 3. are rather convoluted. When i previously did thjs, it wasn't that hard.

Comment: I do find at times the documentation here is a bit brief.

djaus2, 2020-11-25:
I tried https://github.com/Azure/iot-central-firmware/tree/master/tools/dice
I see that has been archived .. does that mean it is no longer correct?
That didn't solve things.

djaus2, 2020-11-25:
I am trying Dave Glovers version of the steps:
https://docs.microsoft.com/en-us/learn/modules/develop-secure-iot-solutions-azure-sphere-iot-central/8-exercise-connect-room-environment-monitor

djaus2, 2020-11-25:
Ok so used Dave's steps, which is very similar to the IoTHub-DPS approach.
So I guess the dice approach IS old hat.

Still getting IOTHUB_CLIENT_CONNECTION_NO_NETWORK  though.
Have doubly checked:
- ScopeID
- TenantID
- App Url

NB: I have seen 2 Component IDs:
- 819255ff-8640-41fd-aea7-f85d34c491d5
- 25025d2c-66da-4448-bae1-ac26fcdd3627

Have tried both here. Where do they come from?
Which is correct for IoTCentral?

bawilless, 2020-11-25:
David,

The component ID’s come from your app_manifest.json file.  The don’t mean much for the exercise you’re doing.  They do come into play when you have applications running on the different MT3620 cores that need to communicate with each other.  They are also used to help identify the application when deploying OTA updates.

Here’s a link to some documentation I wrote on setting up IoT Central for Azure Sphere.  The section starting on page 8 is what you should look at.  The IoT Central interface changes periodically, so it’s difficult to keep documentation current.  The document below was updated Nov 11th, it should be current.

https://avtinc.sharepoint.com/:b:/t/ET-Downloads/EW7fkIP5OP1Hun17d0JmMboB-lUzTiMlXQbrNXzkEcNtvA

Brian

djaus2, 2020-11-25:
Matter resolved:
The  "AllowedConnections": is no longer [ "global.azure-devices-provisioning.net","The app URL"]

It is actual many fold. You generate it according to:

https://docs.microsoft.com/en-us/learn/modules/develop-secure-iot-solutions-azure-sphere-iot-central/8b-exercise-azure-sphere-security

djaus2, 2020-11-25:
Thanks Brian for the clarification.
I've got some blogging to do on this now.

I think Dave's explanation for this is better than what's in the repository.

djaus2, 2020-11-25:
I'll close this tomorrow.

johnnyhli, 2020-11-25:
Thanks @bawilless for the help.
Hi @djaus2 - I'm glad the issues is resolved. The setting up process takes many steps and easily to miss one or two and cause the error. We really appreciate the feedbacks you provided. 

Here are the answers for your questions:

1. Creating CA cert
I believe in [Step 2. Download the tenant authentication CA certificate](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-central#step-2-download-the-tenant-authentication-ca-certificate), you can find the command of creating certificate. 

2. For what should put in AllowedConnections:
In README.md, step [Configure the sample application to work with your Azure Sphere tenant and devices](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTCentral.md#configure-the-sample-application-to-work-with-your-azure-sphere-tenant-and-devices), you can find:
    > IoT Central App URL – this can be found in your browser address bar. For Example: http://myiotcentralapp.azureiotcentral.com/. Paste the URL into the AllowedConnections field of the app manifest.

closed 2020-11-30 by djaus2