Azure/azure-sphere-samples #183: AzureIoT connect to IoT Hub directly readme has missing or incorrect steps

markscott-slalom, 2020-11-26:
The steps for "Connect to IoT Hub directly" in the ReadMe [here ](https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTHub.md) mention that I'll need the Device ID of my Azure Sphere device. The instructions never mention using that value. I am getting the following errors in the output window when trying to run the sample. 

Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_BAD_CREDENTIAL
Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK

Are these errors due to not having the device id in the CmdArgs? Is the Device ID required when using this connection type? If so, how is it added to the CmdArgs? I've tried "--DeviceId", "--DeviceID"  neither which are recognized. 

Please advise. Thank you.

johnnyhli, 2020-11-26:
Hi @markscott-slalom Thanks for the question! I'll look it up.

misaz, 2020-11-26:
There are some secret (probably undocumented) informations what ids and strings must match. The issue you mention is (probably, I am not an MS enginner) caused when Azure refuse your connection for some reason. In my case it was that DeviceId did not match the ID of device registered in IoT Hub. CN of certificate used by device must also match that ID (when using Azure Sphere this is done automaticaly but it is important If you generate private key and signed certificate yourself on non-Azure-Sphere devices). Behaviour in my case (with the same statuses printed as in your case) was that device successfully estabilished TLS connection with valid certificates (both sides accepted connection) but after sending first (encrypted) MQTT packet (I think it was CONNECT packet) server sends me some short encrypted message (I think CONNACK with non-zero status) and then both sides closed the connection. As I mentioned It was caused because I registered device in Azure Portal with my custom name, not a hexadecimal string received using `azsphere dev show-attached` command.

[image]

If you have enough experiences with networking, you can use wireshark to troubleshoot at least unencrypted part (certificate exchange) of communication but for most users it is very hard. I have not found any way how to enforce insecure MQTT connection from the SDK which to troubleshoot MQTT connection process.

Debugging of connection process in this sample costs me about 4 hours. Example itself is very complex and long I think. I like simple short examples rather.

markscott-slalom, 2020-11-27:
@misaz Thanks for the info. I'll check out WireShark. I'm using the Avnet Azure Sphere MT3620 Dev Kit. I did the IoT device registration by hand and followed the instructions [here](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub). I did use the hex value for the device id and not the friendly name.

misaz, 2020-11-27:
So you have step 1, 2 (3 is skipped), 4 and also 5. Correct? Have you set `app_manifest.json` correctly as described in `Connect to the IoT Hub directly` section in https://github.com/Azure/azure-sphere-samples/blob/master/Samples/AzureIoT/IoTHub.md

johnnyhli, 2020-11-27:
@misaz  Thanks for the help.
Hi @markscott-slalom - I think I recreate your error.  Can you double chech with the step 3,4, and 5 in [Authenticate using a direct connection](https://review.docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub?branch=release#authenticate-using-a-direct-connection). Make sure your device is claimed to the tenant that you generated the cert. Thanks!

markscott-slalom, 2020-11-30:
@johnnyhli  I don't have access to that "Authenticate" link you included. I keep getting a 403 Forbidden error.

johnnyhli, 2020-11-30:
Hi @misaz - Sorry for the inconvenience. the link should be https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub#authenticate-using-a-direct-connection.

johnnyhli, 2020-11-30:
Hi @misaze - If the issue resolved, do you have other questions?

markscott-slalom, 2020-11-30:
@johnnyhli My Azure Portal UI shows everything is verified.

markscott-slalom, 2020-11-30:
@johnnyhli Is Device Id a required parameter in the CmgArgs if using the direct connection option? If so, could you please provide an example? Thank you.

johnnyhli, 2020-11-30:
@misaz - But you still can't connect your device, right? is it still showing IOTHUB_CLIENT_CONNECTION_BAD_CREDENTIAL? If so, can you try:
1. claim your device again to the tenant where you create the cert
2. double check the app_manifest.json, especially, the "DeviceAuthentication" field
3. Make sure the device ID you filled in "Create Device" in Azure IoTHub is lower case.

markscott-slalom, 2020-11-30:
@johnnyhli The device id needed to be lower case for the sample to work. It looks like I missed that distinction [here ](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub?branch=release#authenticate-using-a-direct-connection) in step 5. Ironically, the Azure Sphere device spits that value out in upper case.

johnnyhli, 2020-11-30:
@markscott-slalom, you don't need Device ID for CmgArgs. But you do need it for [Step 5. Create an X.509 device for your IoT hub](https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub?branch=release#step-5-create-an-x509-device-for-your-iot-hub)

johnnyhli, 2020-11-30:
@markscott-slalom Glad it works. It is a great feedback. We'll see what we can do here to improve the Customer Experience.

closed 2020-11-30 by johnnyhli