Azure/azure-sphere-samples #185: No ShowIoTCentralConfig.dll

djaus2, 2020-12-06:
```
Samples\AzureIoT\Tools\win-x64> .\showiotcentralconfig
The application to execute does not exist: '<user>\AppData\Local\Temp\.net\ShowIoTCentralConfig\imoz1uwg.5gb\ShowIoTCentralConfig.dll'.
```

I am running the app (or attempting to) in the context of the repository clone.

djaus2, 2020-12-06:
Found an earlier version:  20.1.7.4484  this works.
Current: 20.7.0.26541
???

Nb: I have created a script to automate this step. :)

cawhitworth, 2020-12-07:
If you encounter this error, please delete the `<user>\AppData\Local\Temp\.net\ShowIoTCentralConfig` folder and try again.

ShowIoTCentralConfig is a self-extracting .NET Core executable; by default, it extracts to a folder in Temp. The self-extracting logic assumes that if the folder exists, then it does not need to re-extract - but things in Temp can get cleaned up for all sorts of reasons, so this isn't a safe assumption. It's a bug in .NET Core itself which, I believe, has been fixed in more recent versions. Until we can update and verify this, please continue to delete the folder when you encounter this error.

djaus2, 2020-12-07:
Thanks for that.
Its rather interesting.

David Jones MVP

djaus2, 2020-12-07:
Any chance of getting the source?

David Jones MVP

johnnyhli, 2021-01-15:
Hi @djaus2 - Sorry for the late reply. The ShowIoTCentralConfig  is not open source. I'll close this issue for now. Please let me know if you have any questions. Thanks!

closed 2021-01-15 by johnnyhli