Azure/azure-sphere-samples #186: [IoTCentral] Where to specify our custom device template ID?

m-chau, 2021-01-04:
I am trying to connect Azure Sphere to IoT Central to automatically assign the device to my own device template through DPS, instead of manually migrating the device in the IoT Central application.

However, I cannot find where to specify that piece of information in the sample code or documentation.

johnnyhli, 2021-01-14:
Hi @m-chau, sorry for the late reply. I'll look into it and get back to you soon.

m-chau, 2021-01-25:
@johnnyhli  Thank you! Please keep me updated.

bawilless, 2021-01-25:
Hi Jonny,

I'll jump in and try to help.  There is a way to do what you're asking; this is one of the features that using[ Microsoft's IoT Plug and Play (PnP)](https://docs.microsoft.com/en-us/azure/iot-pnp/overview-iot-plug-and-play) will provide.  

As you've determined IoTCentral needs to know about the device and it's capabilities so that it uses a template that matches the properties.  PnP accomplishes this linkage using the following high level ideas . . . 

- A PnP model is uploaded to the [public PnP repo](https://github.com/Azure/iot-plugandplay-models)
    -  There is a Pull Request Process that you must complete to get your model into the repo
- Your Azure Sphere application must send up some special settings when it connects to the IoT Hub
- Your Azure Sphere application must pass up the model identifier in the IoT Hub connection process
   - This way the PnP compatible cloud solution can locate and pull your model

Once your device connects to IoTCentral as a PnP device and IoTCentral pulls the model from the public repo, it will automatically associate all devices using that model to the PnP Template (that may not be the correct term, but you understand what I'm saying).  You can modify this template with custom views and you're off to the races.

Optionally you can get your device/application [PnP certified](https://docs.microsoft.com/en-us/azure/iot-pnp/howto-certify-device) and included into Microsoft's PnP catalog
- Note that currently the certification portal does not support certification using Azure Sphere devices.  The PnP team is currently working on resolving this constraint.

I have an example that shows how to implement PnP on an Azure Sphere device.  The example is for the Avnet Azure Sphere Starter Kit Out of Box example.  This example is based off Microsoft's AzureIoT example so if you started with that example it should be easy to add PnP support by referencing the example.

- Example Repo: [Link](https://github.com/Avnet/azure-sphere-samples)
- Open the /Samples/AvnetAzureSphereHacksterTTC/HighLevelExampleApp project
- Search the repo for `USE_PNP`

Microsoft has developed tools to help you validate your PnP model and you can use the [AzureIoT Explorer utility](https://docs.microsoft.com/en-us/azure/iot-pnp/howto-use-iot-explorer) (that supports PnP) to test your application and the data that it's sending.

I hope that gets you moving in the right direction.

johnnyhli, 2021-03-01:
Hi @m-chau, I'll close the bug. If you have any questions, please feel free to open this issue. Thanks!

closed 2021-03-01 by johnnyhli