Azure/azure-sphere-samples #192: Docker scripts do not support the latest SDK or Azure Sphere tools version

bawilless, 2021-02-02:
I updated my repo to pull the latest MSFT changes and I was excited to see the .devcontainer folder at the top level of the repo.  I copied this folder into one of my projects, opened the project, allowed the docker container to build and then tried to build my project.  My project has been updated to support the latest 21.01 tools and TARGET_API_SET.  

After the docker container built, I was prompted to perform a reconfigure operation.  When I did the console reported a tools version error.

[cmake] CMake Error at /opt/azurespheresdk/CMakeFiles/AzureSphere.cmake:36 (message):
[cmake]   azsphere_configure_tools: unsupported tools version "21.01".  Only
[cmake]   ['20.04', '20.07', '20.10'] are supported.
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:7 (azsphere_configure_tools)

When I updated my CMakeLists.txt file to roll back the TOOLS_REISION to "20.10" and the TARGET_API_SET to "7", my project configured correctly and built and ran.

This looks like the docker build script is not installing the latest SDK 21.01.

Environment: Ubuntu 20.04.1 LTS
IDE: Visual Studio Code

jophippe, 2021-02-03:
I cannot reproduce this issue. We pushed the 21.01 Docker container on Monday afternoon, so if you ran the dev container before that, you may have pulled the 20.10 container instead. If that's the case, can you try running the dev container again and see if it works? If not, try just running the latest Docker container manually and run "/opt/azurespheresdk/Tools/azsphere show-version" in it to make sure the version is 21.01

bawilless, 2021-02-05:
I started to dig into this issue this morning and realized that I was trying to use the docker container in a way that it was not designed for.  Closing this issue.

closed 2021-02-05 by bawilless