Azure/azure-sphere-samples #195: Data Sharing between Click Boards

nishithvpoojary, 2021-02-19:
Is there any code sample to transfer data from Click1 to Click2. I want to transfer the Click1 Rx data to Click2 and send it to PC using RTOS.

johnnyhli, 2021-02-19:
Hi @nishithvpoojary - From [Azure Sphere MT3620 Starter Kit Hardware User Guide](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjZto2Lv_buAhUVrJ4KHaTRDzEQFjAAegQIBRAD&url=https%3A%2F%2Fwww.avnet.com%2Fopasdata%2Fd120001%2Fmedias%2Fdocus%2F196%2FAzure%2520Sphere%2520Starter%2520Kit%2520User%2520Guide%2520(v1.3).pdf&usg=AOvVaw14g29ZL8DJZxedDWTFw1P2), Page 19, you can find that the I2C on both Click1 and Click2 share the same GPIO. The following samples might help you: The Azure Sphere samples of  a bare-metal [HelloWorld sample](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/HelloWorld/HelloWorld_RTApp_MT3620_BareMetal), the sample that demonstrates[ inter-core communication](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/IntercoreComms/IntercoreComms_RTApp_MT3620_BareMetal) between high-level and RTApps, as well as the sample that shows how to use [Azure Sphere with Azure RTOS](https://github.com/Azure-Samples/Azure-RTOS-on-Azure-Sphere-Mediatek-MT3620). 

Please let me know if you have further questions. Thanks.

nishithvpoojary, 2021-02-20:
Does that mean, If I set "ISU0" in app.manifest.json and send a "Hello" message,  this message is sent to both Click1 and Click2 ??

johnnyhli, 2021-02-22:
Hi @nishithvpoojary - Thanks for the clarification. I will look more into your questions. For the question of displaying receiving data, is it OK to display in VS Code or Visual Studio Device Output window during debugging? If yes, have you check out [UART sample code](https://github.com/Azure/azure-sphere-samples/tree/master/Samples/UART)? Currently we don't have sample code for FreeRTOS. But we do have sample that shows how to use [Azure Sphere with Azure RTOS](https://github.com/Azure-Samples/Azure-RTOS-on-Azure-Sphere-Mediatek-MT3620).

nishithvpoojary, 2021-02-23:
Hi. Johnnyhli. Thanks for your response . I was able to transfer data from Click1 to Click2 and was able to display the data from Click2 using Terminal. I had to make a little bit of changes from the FreeRTOS UART code. 
Thank you.
Is there any BareMetal UART code?

closed 2021-02-23 by nishithvpoojary