Azure/azure-sphere-samples #197: IoTHubDeviceClient_LL_UploadToBlob not availaible

misaz, 2021-03-09:
Sample https://github.com/Azure/azure-sphere-samples/tree/master/Samples/AzureIoT shows how to use IoTHubDeviceClient. I need to modify this example to upload file from device to blob storage, but function IoTHubDeviceClient_LL_UploadToBlob is not avalaible on Azure Sphere. Compilation crash and it is also not present in any header in `C:\Program Files (x86)\Microsoft Azure Sphere SDK\Sysroots\8\usr\include`.

Can I upload file to azure using SDK or I have to implement it myself?

SatishBoddu-MSFT, 2021-03-17:
Similar: https://docs.microsoft.com/en-us/answers/questions/306549/index.html

johnnyhli, 2021-03-17:
Hi @misaz - Thanks for asking the question. We are also looking into your question [here](https://docs.microsoft.com/en-us/answers/questions/306549/index.html).  

Thanks @SatishBoddu-MSFT for the help!

tokenstar, 2021-07-05:
closing this issue as this is not an issue with the samples per se, but a feature request for the Azure IoT SDK included in the Azure Sphere OS.  Thanks @misaz  for the input.

closed 2021-07-05 by tokenstar