Azure/azure-sphere-samples #198: Azure device not responding error

nishithvpoojary, 2021-03-17:
Hello,
I was working on PrivateNetworkServices sample provided by the Azure from this link (https://github.com/Azure/azure-sphere-samples/tree/master/Samples/PrivateNetworkServices).
I was able to get output on Command prompt when I used WiFi(“wlan0”). But when I changed the Network Interface to “eth0” in the ‘main.c’ program, I get an error in Visual studio as “The device is not responding. The device may be unresponsive if it is applying an Azure Sphere operating system update; please retry in a few minutes. (DeviceConnectionException)” and an error in Windows command prompt as below pic. 

[image]
 
I have tried to figure it out by following the Microsoft document, but still didn’t work. So I did the Device Recovery using the command “azsphere device recover”. The error was gone until I flashed the same program again for ethernet connection.  I thought it might be device problem and so I changed the Azure device. I got the same error again at the exact moment when I made the Network Interface changes. How is this possible? Please guide me with this. Thank You.

bawilless, 2021-03-17:
I've never seen this issue before, here are a couple of thoughts . . . .

I'll assume you're using one of the Avnet Azure Sphere Starter Kits

When you change the application to use eth0 . . . 
1. Make sure you follow the detailed instructions [here](https://docs.microsoft.com/en-us/azure-sphere/network/connect-ethernet?tabs=cliv2beta) to prepare your device to use the add on ethernet hardware
2. Verify that you are using an Avnet Starter Kit Rev2 (Rev1 does not support the lan-enc28j60-isu0-int5 board configuration package) The revision is printed on the PCB under the Avnet Azure Sphere Module.
3. Make sure you insert the [ETH Click board](https://www.mikroe.com/eth-click) into Click Socket #1 on the starter kit

Since your device can communicate to the development computer when you don't have your ethernet application installed, lets make sure your ethernet application is configured correctly first.  

I work for Avnet and I'm familiar with the Avnet Starter Kits

nishithvpoojary, 2021-03-17:
Thank you for your reply Bawilless. I use avnet Starter Kit Rev2 board. I have followed the "Connect to Azure Ethernet" Microsoft document thoroughly. I have also inserted the ETH click board in the Click Socket 1 on the kit.

nishithvpoojary, 2021-03-17:
I used the command command “azsphere device recover” now and I deleted the Ethernet image package and created it again. Then I sideloaded the application and ethernet image package. Now I have a different error in Visual Studio.
[image]

nishithvpoojary, 2021-03-17:
Now again I have the same errors. 
[image]
Please tell me what I have to do.
I used the same Ethernet image package as given in the document.
[image]

bawilless, 2021-03-17:
I'm trying to reproduce your issue and so far I don't see the same result as you.  Here's what I did . . . 

* Using an Avnet Starter Kit Rev 2 board with Eth Click installed in Click socket 1
* Eth Click connected to a live ethernet port (link light on my ETH Click is on)

1. Recovered my device 
    `azsphere device recover`
2. Enabled development mode 
    `azsphere device enable-development`
3. Downloaded the latest board configuration file 
    `azsphere image-package pack-board-config --preset lan-enc28j60-isu0-int5 --destination enc28j60-isu0-int5.imagepackage`
4. Side loaded the board configuration package onto my device: 
    `azsphere device sideload deploy --image-package enc28j60-isu0-int5.imagepackage`
5.  Pulled the latest example repo from MSFT 
    `git clone https://github.com/Azure/azure-sphere-samples.git`
6. Verified I can communicate with my device 
    `azsphere device show-attached`
7. Load the Private Networking Services example in Visual Studio 2019
8. Run the example without any changes
9. Example runs . . . 

    Remote debugging from host 192.168.35.1, port 54939
    INFO: Private Ethernet TCP server application starting.
    INFO: Networking_GetInterfaceCount: count=2
    INFO: Networking_GetInterfaces: actualCount=2
    INFO: interface #0
    INFO:   interfaceName="wlan0"
    INFO:   isEnabled="1"
    INFO:   ipConfigurationType=0 (DhcpNone)
    INFO:   interfaceMediumType=1 (Wi-Fi)
    INFO:   interfaceStatus=0x07
    INFO: interface #1
    INFO:   interfaceName="eth0"
    INFO:   isEnabled="1"
    INFO:   ipConfigurationType=0 (DhcpNone)
    INFO:   interfaceMediumType=2 (Ethernet)
    INFO:   interfaceStatus=0x07
    INFO: Set static IP address on network interface: eth0.
    INFO: SNTP server has started on network interface: eth0.
    INFO: DHCP server has started on network interface: eth0.
    INFO: TCP server: Listening for client connection (fd 7).

Question: Are you using the latest example application?  My example is using eth0 by default.

10. Verified I can communicate with my device 
    `azsphere device show-attached`
11. Following the instruction from the example, change the application to use wlan0
12. Run the application . . . 

    Remote debugging from host 192.168.35.1, port 55795
    INFO: Private Ethernet TCP server application starting.
    INFO: Networking_GetInterfaceCount: count=2
    INFO: Networking_GetInterfaces: actualCount=2
    INFO: interface #0
    INFO:   interfaceName="wlan0"
    INFO:   isEnabled="1"
    INFO:   ipConfigurationType=0 (DhcpNone)
    INFO:   interfaceMediumType=1 (Wi-Fi)
    INFO:   interfaceStatus=0x07
    INFO: interface #1
    INFO:   interfaceName="eth0"
    INFO:   isEnabled="1"
    INFO:   ipConfigurationType=0 (DhcpNone)
    INFO:   interfaceMediumType=2 (Ethernet)
    INFO:   interfaceStatus=0x07
    INFO: Set static IP address on network interface: wlan0.
    INFO: SNTP server has started on network interface: wlan0.
    ERROR: Networking_DhcpServer_Start: 1 (Operation not permitted)
    INFO: Application exiting.

    Child exited with status 8

13. I'm looking into this error . . . 
14. Verified I can communicate with my device 
    `azsphere device show-attached`
15. I disconnected the LAN cable and can still communicate with the device

I can't reproduce your issue.  Can you pull the latest repo and try to follow my steps to see if you still have the error?

nishithvpoojary, 2021-03-17:
I have followed the same steps. I used the program from the following link which has eth0 as default from this link,
"https://github.com/Azure/azure-sphere-samples/tree/master/Samples/PrivateNetworkServices"
I got this error
[image]

bawilless, 2021-03-17:
The initial issue you reported was that your development PC could not communicate with your board when your application used eth0, are you still having that issue?

nishithvpoojary, 2021-03-17:
I used the step 1 command. So it worked. Now I have this error below.
[image]

bawilless, 2021-03-17:
I can't find a way to reproduce your issue on my system.  Sorry I'm not an expert on this example or why you're seeing that error.  At least I helped you past the initial issue.  

You should consider opening a new issue on the new problem [here](https://docs.microsoft.com/en-us/answers/search.html?c=&includeChildren=&f=&type=question+OR+idea+OR+kbentry+OR+answer+OR+topic+OR+user&redirect=search%2Fsearch&sort=relevance&q=Azure%20Sphere).  That forum will get more visibility from MSFT Azure Sphere experts and from the community as well.  Make sure you tag your question with "Azure Sphere" so it will be seen by the Azure Sphere experts.

nishithvpoojary, 2021-03-17:
Ok. Thank you very much.

johnnyhli, 2021-03-17:
Hi @bawilless - Thanks for your help!
Hi @nishithvpoojary - Thanks for your questions. do you mind let me know more information so that I can better help you? Please send followings to AzCommunity@microsoft.com (do not attach it here):
1. A link back to this issue
2. Support Data by following [this instruction](https://docs.microsoft.com/en-us/azure-sphere/reference/azsphere-get-support-data?tabs=cliv2beta) 
3. main.c and app_manifest.json in your Private Ethernet sample.

Thank you so much!

nishithvpoojary, 2021-03-17:
Thank You johnnyhli. I have sent it.

johnnyhli, 2021-03-17:
Hi @nishithvpoojary - it seems you main.c still showing that you set wlan0 in `static const char NetworkInterface[];`  
[image]

nishithvpoojary, 2021-03-18:
That I changed it after I got the error. This below pic was the output with network interface as "wlan0".
[image]
The moment I change it to "eth0", I get an error.

asergaz, 2021-03-19:
linking same question asked in MicrosoftQ Q&A: https://docs.microsoft.com/en-us/answers/questions/318686/index.html

@nishithvpoojary please can you Verify Chandra answer? Thanks

nishithvpoojary, 2021-04-08:
I still have this error. 
[image]
Why does it shows failed for 'eth0'?

chandran-Az, 2021-05-07:
Closing this thread as it was found the chipset W5500 is not supported with Azure Sphere.

closed 2021-05-07 by chandran-Az