Azure/azure-sphere-samples #199: Visual Studio launch.vs.json Error

Prathik-Bhundary, 2021-04-01:
Hello. I use Avnet Azure Sphere Starter kit Revision 2 Board and I use Visual Studio 2019 Enterprise.
 I have 3 Warnings without any errors. There are my warnings below.
[image]

I have green lines in my launch.vs.json program
[image]
 
How can I fix this? Thank You

bawilless, 2021-04-01:
Which example application are you trying to build?

Prathik-Bhundary, 2021-04-01:
UART sample High Level Application

bawilless, 2021-04-01:
I can't reproduce your issue using Visual Studio 2019 V16.9.2.  Make sure . . . 

1. You clone the entire repo: https://github.com/Azure/azure-sphere-samples.git
2. To use the Avnet Starter Kit Rev 2 make sure the CMakeLists.txt file contains the following line:

[image]

When using Visual Studio Code, I often see warnings that go away once I actually do a build.

Prathik-Bhundary, 2021-04-01:
I too have set the same Target Hardware. And I can build without any problem. It shows Build all succeeded. But when I open the launch.vs.json , it shows this error

bawilless, 2021-04-01:
Are you running the latest Azure Sphere Visual Studio Extension?

[image]

bawilless, 2021-04-01:
I've also see a fresh CMake Generate Cache operation clean up things like this  . . . .

Right click on CMakeLists.txt and regenerate the cache

[image]

Prathik-Bhundary, 2021-04-01:
[image]

I too clicked the generate option
[image]

Still I have the error

bawilless, 2021-04-01:
Sorry, I don't think I can be much help at this point.  It seems like this is a tools/configuration issue, since I can't reproduce the issue on my system using the same source project.  I recommend that you post a comment on the Microsoft Q&A site.  Your issue will be seen by more of the Microsoft support staff then here.  The link is: https://docs.microsoft.com/en-us/answers/topics/azure-sphere.html

robotdad, 2021-04-01:
You may need to delete the cache before regeneration.
[image]

Prathik-Bhundary, 2021-04-02:
I deleted the cache. Now I have 1 error and 3 warnings
[image]

Prathik-Bhundary, 2021-04-05:
Its not causing any problem while building or debugging. So I will ignore it and continue as it is. I will close this issue. Thank You.

closed 2021-04-05 by Prathik-Bhundary