Azure/azure-sphere-samples #206: Updated AzureIoT example does not re-provision/re-connect when the device is deleted from the IoTHub

bawilless, 2021-05-12:
The most recent version of the AzureIoT example does not handle the case where a device is deleted from the IoTHub while the device is connected.  In the previous implementation this case was handled, and the application would re-provision/connect to the IoTHub

### Desired behavior:

The application should re-provision/re-connect to the IoTHub without requiring an application restart.

### Steps to reproduce:
1. Configure the application to connect to a IoTHub (I used the DPS configuration)
2. Run the application and verify that it's connected to the IoTHub
3. Delete the device from the IoTHub (I did this from the Azure IoT Explorer utility)
4. Observe that the application never re-provisions/re-connects

### Debug from previous implementation (working case): (I removed irrelevant debug for clarity)
    ConnectionType: DPS
    ScopeID: 0ne0008xxxx
    Using DPS Connection: Azure IoT DPS Scope ID 0ne0008xxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_OK'.
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_OK

    // Delete device from IoTHub
 
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK
    IoTHubDeviceClient_LL_CreateWithAzureSphereDeviceAuthProvisioning returned 'AZURE_SPHERE_PROV_RESULT_OK'.
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_OK

### Debug from the current implementation: (I removed irrelevant debug for clarity)
    ScopeID: 0ne0008xxxx
    Using DPS Connection: Azure IoT DPS Scope ID 0ne0008xxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    INFO: Azure IoT Hub connection started.
    INFO: DPS device registration successful
    INFO: Azure IoT Hub connection complete.
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_OK

    // Delete device from IoTHub

    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK
    INFO: Azure IoT Hub connection started.
    INFO: Azure IoT Hub connection complete.
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_BAD_CREDENTIAL
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    [Azure IoT] Using HSM cert at INFO: Azure IoT Hub connection started.
    INFO: Azure IoT Hub connection complete.

    // Repeats forever . . .

    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_BAD_CREDENTIAL
    INFO: Azure IoT Hub connection started.
    /run/daa/8d34f65c-532e-4dcf-a1d6-xxxxxxxxxxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-xxxxxxxxxxxxxxxxxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-xxxxxxxxxxxxxxxxxxx
    [Azure IoT] Using HSM cert at /run/daa/8d34f65c-532e-4dcf-xxxxxxxxxxxxxxxxxxx
    INFO: Azure IoT Hub connection complete.
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_NO_NETWORK
    Azure IoT connection status: IOTHUB_CLIENT_CONNECTION_BAD_CREDENTIAL
    INFO: Azure IoT Hub connection started.
    INFO: Azure IoT Hub connection complete.

tokenstar, 2021-05-13:
Thanks for the issue report Brian, we're looking into this.

closed 2021-05-14 by cawhitworth

bawilless, 2021-05-14:
Thanks for the quick turn on this fix Chris.  I've validated it on my side as well.

cawhitworth, 2021-05-14:
Thanks also due to @mikehall-ms for diagnosing.

PR #207 (2021-05-14): Ensure DPS provisioning state is set to false/invalid before starting
Fix #206 

Provisioning state was not reset when starting provisioning, meaning subsequent disconnects/reprovisions did not happen correctly.