Azure/azure-sphere-samples #210: Azure Sphere device restart

nishithvpoojary, 2021-06-01:
I use Avnet Azure sphere starter kit rev 2 board.
How can I restart the device without using commands or device reset button. Is there any way to do it using Rest API? I want to update the application package without touching the device. But it checks for updates only when resetting the device. Please let me know how I can do it.

Thank you.

chandran-Az, 2021-06-01:
Hi Nishith - Azure Sphere can be rebooted using the power management functions documented here https://docs.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-power/function-power-forcesystemreboot

The powerdown sample covers the scenario of shutting down the device and waking up after a certain interval - https://github.com/Azure/azure-sphere-samples/tree/master/Samples/Powerdown/Powerdown_HighLevelApp

Devices check for updates every 24 hours or when they are rebooted. Currently, there's no command to force update check.

nishithvpoojary, 2021-06-08:
Issue Solved.

[https://docs.microsoft.com/en-us/answers/questions/417820/azure-sphere-device-reset.html](url)

closed 2021-06-08 by nishithvpoojary