Azure/azure-sphere-samples #215: External MCU application update not working

nishithvpoojary, 2021-09-01:
Hello,
I use Avnet Starter kit Rev2 and nrf52840 development board.
I used the "HelloWorld" high level application and added the External MCU update code here.
I was able to send the dfu softdevice but causes issue while sending the dfu application. I get a bootloader issue.
[image]

Please help me out with this issue.

bawilless, 2021-09-01:
Were you able to run the https://github.com/Azure/azure-sphere-samples/tree/master/Samples/ExternalMcuUpdate example on the Starter Kit Rev2 without any modifications?

nishithvpoojary, 2021-09-01:
> Were you able to run the https://github.com/Azure/azure-sphere-samples/tree/master/Samples/ExternalMcuUpdate example on the Starter Kit Rev2 without any modifications?

I had make few modifications. Used softdevice140 instead of s132. Changed hardware settings required for Avnet rev2 board.
Removed nRFReset  by pin functions and added json data which is sent to nrf52. When the nrf52 receives this json data, the device goes to bootloader mode.

nishithvpoojary, 2021-09-17:
What should I do for this softdevice issue?

gtrevi, 2022-01-11:
@nishithvpoojary 
This issue has not much relation to the Azure Sphere updating sample itself, since the softdevice upgrade validation is done on the nRF bootloader (`nrf_dfu_validation.c` & `nrf_dfu_ver_validation.c`). Clearly, if is not supplied in the update batch, the problem does not arise.

S140 should be in fact compatible with the nRF52840 SoC (see [here](https://www.nordicsemi.com/Products/Development-software/S140)), so I suggest checking out on the nRF documentation and bootloader source code to verify the available options.
Also, the bootloader itself should emit debug logs, from where you can check the code-paths taken for the softdevice validation that is failing (see [here](https://devzone.nordicsemi.com/f/nordic-q-a/16979/where-can-i-see-nrf_log-output)).

gtrevi, 2022-01-21:
Closing this. Feel free to follow-up in case needed.

closed 2022-01-21 by gtrevi