Azure/azure-sphere-samples #216: CMake rule to auto-update hardware definition header files?

henrygab, 2021-09-09:
### Request:

Update the `azsphere_target_hardware_definition()` function to add rules necessary to ensure that the `.h` file matches the `.json` board definitions.  This appears possible by using `azsphere hardware-definition` command as a custom command.

### Why this is useful

### Keep magic smoke in the chip

This causes problems when modifying the pin being referred to in `foo.json`, without adding any newly defined names.  The resulting error (use of wrong pin) is entirely **_silent_**, until you cause the magic smoke to exit the chip, resulting in a long hardware-debug session where everything is confirmed to be just fine.  This is significant pain for something that CMake is designed to prevent ... and can prevent hardware destruction.

#### Improve initial user experience expanding from simple samples.

Even if just wanting to modify `blink` sample to blink another LEDs (e.g., the RGB for both v1 and v2 AVNET kits),  modifying the hardware definition files in the project correctly, and then attempting to build ... results in failure.  The error complains that the newly-defined hardware resources are not defined.

While this is true, it is only true because CMake wasn't configured to track the dependency between the `foo.json` and the `hw/inc/foo.h` file, and to regenerate the header when the source json is newer.

### Example workaround / hack

<details><summary>My skills with CMake leave much to be desired, but click to expand</summary><P/>

I tried the following, which appears to work, regenerating the header only when it changes.

```CMake

# Put this is near top of file, near TARGET_HARDWARE and TARGET_DEFINITION
set(TARGET_HEADER     "inc/hw/template_appliance.h")

# modify the add_executable line to depend on the header
add_executable (
    ${PROJECT_NAME}
    main.c
    "${TARGET_DIRECTORY}/${TARGET_HEADER}"
    )

# Create a target to always ensure header is updated?
ADD_CUSTOM_TARGET(
    HW_HEADER ALL
    DEPENDS "${TARGET_DIRECTORY}/${TARGET_DEFINITION}"
    )

# Add the custom command that generates the header
add_custom_command(
    OUTPUT
        "${TARGET_DIRECTORY}/${TARGET_HEADER}"
    COMMAND
        azsphere
        hardware-definition
        generate-header
        --hardware-definition-file "${TARGET_DIRECTORY}/${TARGET_DEFINITION}"
    DEPENDS
        "${TARGET_DIRECTORY}/${TARGET_DEFINITION}"
    WORKING_DIRECTORY
        ${CMAKE_CURRENT_SOURCE_DIR}
    COMMENT
        "Ensure the generated hardware header file matches the JSON"
    VERBATIM
    )
```

</details><HR/>

I make no claim that this hack is a "right" solution, but hopefully it's a good starting point.

tokenstar, 2021-09-09:
Thanks for the issue!  This was designed this way because often .json/.h files are in shared-project or SDK locations and having individual projects rebuild .json/.h files in shared locations has issues.  But your pain point is totally valid and we'll have a think about how to address that.  Any further feedback or thoughts welcome!

henrygab, 2021-09-10:
Yes, I thought that could be one issue ... but even then...

#### Isn't it be considered "good" to have the build **_at least_** fail, if the headers and json don't match?

OK, presuming that it would be good, the question becomes "how"?

The exit code from the following command indicates whether the json and header are known to match:
`azsphere  hardware-definition test-header --hardware-definition-file foo.json`

The `exit code` is zero when the headers match, and the `exit code` is non-zero when there is a problem. (`2` is used for mismatches, for when the JSON file is not found, and potentially for other errors....)

<details><summary>Consider the following solution (click to expand)</summary><P/>

1. Have a list of hardware definition files (or, root directories) for those shipped with the SDK. ... this probably already exists, or is trivial to generate.  This goes into the SDK.

2. Update the function `azsphere_target_hardware_definition()` to register the custom command and custom target for the current build's hardware definition files (see above example ... would go into the SDK)

3. Within the project's `CMakeList.txt`, define a separate, user-specified list of directory roots that store the user-managed hardware definition directories (already somewhat exists ... see [`find_path()`](https://github.com/Azure/azure-sphere-samples/blob/8cc5ed5cceafad21bf7a1eece76ddfa41d2b0687/Samples/GPIO/GPIO_HighLevelApp/CMakeLists.txt#L28) ... just put that list into a variable)

4. Rather than unconditionally generating the `.h` file when it's older (as my workaround does)...
    a. First run the `test-header` subcommmand
    b. Continue build if the `LastExitCode` / `%ERRORLEVEL%` is zero (no update required)
    c. Else if ***`[[CONDITIONALS?]]`***, attempt to re-generate the header file (and continue these steps)
    d. Run the `test-header` subcommand again
    e. Continue build if the `LastExitCode` / `%ERRORLEVEL%` is zero (no update required)
    f. Else, fail the build with appropriately helpful error message

Where ***`[[CONDITIONALS?]]`*** is still an open question, but some options include:
* Only attempt to automatically regenerate files that exist under the current project directory
* Attempt to automatically regenerate files, unless they exist under one of the SDK directories
* Attempt to automatically regenerate files, only if they exist under one of the user-specified directories
* NEVER attempt to automatically regenerate the files

In all cases, when the build fails due to a mismatch, it would be useful to have a corresponding task (build target?), whose sole purpose is to unconditionally update the header files to match the JSON files.  This would essentially be identical to the steps in #4, above.

</details>

This would seem to be an appropriate use of `CMake` and `Ninja` ... ensuring that generated files are not out of date, right?

tokenstar, 2022-02-07:
Thanks @henrygab  for the feature suggestion for the SDK's cmake files.  I have copied this into an internal backlog item.