Azure/azure-sphere-samples #222: Azure sphere External device firmware update issue

nishithvpoojary, 2022-01-08:
I use Avnet Starter kit Rev2 board. I followed the External firmware update code as provided in the Sample.

[1]: https://github.com/Azure/azure-sphere-samples/tree/master/Samples/ExternalMcuUpdate

I was able to send the firmware to the nrf52 device without enabling the CTS/RTS on Avnet device, but it is enabled on the nrf52 device.

Conditions are as belows:-
1) If Azure CTS/RTS is disabled and nrf CTS/RTS is enabled, and no jumper wires are connected between nrf CTS/RTS to azure RTS/CTS, firmware update won't work.
2) If Azure CTS/RTS is disabled and nrf CTS/RTS is enabled, and if a jumper wire is connected between Azure CTS to nrf52 RTS, then the firmware update will work.

How could this work. Please help me with this working logic. Thank you.

asergaz, 2022-01-10:
Relates with: https://docs.microsoft.com/en-us/answers/questions/669470/index.html

gtrevi, 2022-01-11:
@nishithvpoojary
As a general rule, it's recommended that UART flow control configurations do not differ on the two communication parties.

That said, the likely reason for which case 2) works is that when the Azure Sphere's CTS pin is connected to the NFR's RTS, its current state (undefined btw) "happens" to trigger the hardware flow control circuit between the two.
To prove this, you can use case 1), and simply try to separately pull-up(3.3V)/down(GND) the NRF's RTS only: the FW update should similarly succeed/fail respectively.

Therefore, you can either:

1. Enable flow control on Azure Sphere (on the NRF52 is already enabled in its bootloader) **and** cross-connect CTS/RTS between the two (like shown in the sample).
1. Disable flow control on Azure Sphere **and** pull-up RTS (and CTS optionally, since already done by the bootloader) on the NRF... but accept potential communication issues.

Hope this helps.

gtrevi, 2022-01-21:
Closing this. Feel free to follow-up in case needed.

closed 2022-01-21 by gtrevi

nishithvpoojary, 2022-01-28:
Thank you for your answer gtrevi. 
Can I change the flowcontrol for the UART if needed?

gtrevi, 2022-01-31:
Sure, you can fully configure the UART with the `UART_Config` structure before opening, please check out the following doc:

https://docs.microsoft.com/azure-sphere/reference/applibs-reference/applibs-uart/struct-uart-config

nishithvpoojary, 2022-02-07:
What I meant was that, If I set the UART flow control as "None" at the beginning , Can I change the Flowcontrol to CTS/RTS only when I need it.?
Thank you

gtrevi, 2022-02-09:
Yes, at runtime you can close the UART fd and reopen the peripheral with different configuration settings.

gtrevi, 2022-02-09:
Yes, at runtime you can close the UART's fd and reopen the peripheral with different configuration settings.