Azure/azure-sphere-samples #230: Azure sphere Hub authentication issue

nishithvpoojary, 2022-03-15:
Hi. I followed the Azure IOT sample on Avnet starter kit. I have two Hub available. I could authenticate and get data using one hub but I couldn't connect to another Hub. I have checked the Scope Id and the IOT Hub hostname. Its properly written in the app_manifest.json . I have also made sure that the hub is linked to dps and that the hub hostname is selected in the enrollment group.
Please refer the image for the error obtained.
[image]

Thank you.

bawilless, 2022-03-15:
Are you using the same DPS for both hubs?  The DPS will return the IoTHub for the device to use, make sure your DPS::Enrollment group correctly lists the hub you want your device to use.  If the original enrollment group is still defined, you may need to delete that one and add a new one that specifies your new IoTHub, or just change the IoTHub that it uses.  

The DPS will select which enrollment group to use based on the devices CA certificate (that's how I think of it anyway).  I'm not sure what the behavior is when there are multiple enrollment groups that list the same certificate, but I would make sure you only have one enrollment group defined for your CA certificate.

You can test this theory by including the original IoTHub hostname in your "allowedConnections" list.  If it connects, then you know that the DPS is returning the wrong IoTHub and you need to clean up the enrollment groups.

nishithvpoojary, 2022-03-15:
Thank you. It worked. The reason it didn't work because the hub hostname was selected in other enrollment group as well. When I selected the hub hostname on only one enrollment group and deselected from other groups, it worked. Thanks a lot bawilless.

closed 2022-03-15 by chandran-Az