Azure/azure-sphere-samples #233: Device is not responding when deploying

Martin1509, 2022-03-23:
I have problem with deploying azure sphere sample to the device.
It was working fine, but after I updated azure sphere sdk to the newest version 22.02.3.41775, it starts to show error when deploying. But I'm sure its related to different issue then updated sdk because I installed this version of sdk to another computer and it worked fine.

I can also recover device from CMD and set wifi settings for it. But I cannot deploy anything. 

**Error from deploying**

Deploying to the device with an ID starting with 'cannot g' at '192.168.35.2'.
azsphere device sideload delete -e dc7f135c-6074-4d49-aa3a-160e4eed884f -ip 192.168.35.2 has exited with code 2.

error: The device is not responding. The device may be unresponsive if it is applying an Azure Sphere operating system update; please retry in a few minutes.

error: The device is not responding. The device may be unresponsive if it is applying an Azure Sphere operating system update; please retry in a few minutes.

**VS2019 Azure Sphere Explorer**

[image]

**CMD is working**

`azsphere device show-attached`
*Device ID: ..*
*IP address: 192.168.35.2*
*Connection path: 113*

`azsphere device wifi show-status` shows that device is properly connected.

**I tried to**

- reinstall Azure Sphere SDK
- restart the Azure Sphere Device Communication Service
- reinstall of drivers for USB serial converters
- check TAP-Windows adapter
- check Cisco AnyConnect VPN client
- remove config file
- run `azsphere device recover`

tokenstar, 2022-03-23:
Hi @Martin1509  - thanks for the issue. Please can you report the version number of the "Azure Sphere for Visual Studio" extension you are using, and please confirm whether you have tried restarting Visual Studio.

Martin1509, 2022-03-23:
Hi @tokenstar. Version number is 22.2.3.33494. I updated VS2019 and also tried to repair it.

olegkhar, 2022-03-23:
Hi @Martin1509
Sorry that you are experiencing these issues.  To help us diagnose the behavior in VS Extension, can you please set the environment variable "AzureSphere.PrintCliCommands" to 1, and restart Visual Studio?  The output of AzSphere CLI will be shown in the Output window; it will help us understand the issues.

Thanks!

olegkhar, 2022-03-23:
@Martin1509 
Thanks for the prompt response!  I'm trying to narrow the issue down to either the Visual Studio environment or the AzSphere CLI.  Can you please run 3 more commands:

1. In PowerShell, please run `azsphere show-user -o json` and `azsphere device list-attached`
2. In "Azure Sphere Classic Development Command Prompt (Deprecated)", can you please run `azsphere device list-attached`

Thank you!

Martin1509, 2022-03-23:
@olegkhar 

`azsphere show-user -o json`

The command failed with an unexpected error. Here is the traceback:
No section: 'sphere'
Traceback (most recent call last):
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\knack/cli.py", line 231, in invoke
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/cli/core/commands/__init__.py", line 657, in execute
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/cli/core/commands/__init__.py", line 720, in _run_jobs_serially
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/cli/core/commands/__init__.py", line 691, in _run_job
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/cli/core/commands/__init__.py", line 328, in __call__
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/cli/core/commands/command_operation.py", line 121, in handler
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/sphere/cli/core/profile.py", line 239, in get_user
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/sphere/cli/core/config.py", line 144, in get_azsphere_config_value
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\azure/sphere/cli/core/config.py", line 114, in get_global_config_value
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\knack/config.py", line 97, in get
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\knack/config.py", line 92, in get
  File "D:\a\_work\1\s\azsphere-cli\build_scripts\windows\artifacts\cli\Lib\site-packages\knack/config.py", line 206, in get
  File "configparser.py", line 781, in get
  File "configparser.py", line 1149, in _unify_values
configparser.NoSectionError: No section: 'sphere'

`azsphere device list-attached`

1 device attached:
--> Device ID: eaddb5...
  --> Is responsive: yes
  --> IP address: 192.168.35.2
  --> Connection path: 113

`azsphere device list-attached` deprecated

1 device attached:
error: The device is not responding. The device may be unresponsive if it is applying an Azure Sphere operating system update; please retry in a few minutes.
--> Device ID: Cannot get device ID
  --> Is responsive: no
  --> IP address: 192.168.35.2
  --> Connection path: 113

Martin1509, 2022-03-23:
@olegkhar 
Maybe this information will help too.

[image]

Martin1509, 2022-03-24:
@olegkhar @tokenstar 

**Azure Sphere Classic developer Command Prompt (Deprecated)** `azsphere device recover -v -diag`

```
verbose: Azure Sphere Utility version 22.2.3.41775
verbose: Copyright (C) Microsoft Corporation. All rights reserved.
verbose:
verbose: Start time (UTC): Thursday, 24 March 2022 10:20:42
Downloading recovery images...
verbose: Downloading https://prod.releases.sphere.azure.net/recovery/mt3620an.zip to C:\Users\marti\AppData\Local\Temp\AzureSphereRecoveryImages\77d2823e-fae1-4809-9f06-a5809932e454\mt3620an.zip.
error: Device recovery failed: Unable to download recovery images for chip 'mt3620an'. Check your network connection.
trace: System.Net.WebException: Unable to connect to the remote server ---> System.Net.Sockets.SocketException: No connection could be made because the target machine actively refused it 127.0.0.1:8888
trace:    at System.Net.Sockets.Socket.InternalEndConnect(IAsyncResult asyncResult)
trace:    at System.Net.Sockets.Socket.EndConnect(IAsyncResult asyncResult)
trace:    at System.Net.ServicePoint.ConnectSocketInternal(Boolean connectFailure, Socket s4, Socket s6, Socket& socket, IPAddress& address, ConnectSocketState state, IAsyncResult asyncResult, Exception& exception)
trace:    --- End of inner exception stack trace ---
trace:    at System.Net.HttpWebRequest.EndGetResponse(IAsyncResult asyncResult)
trace:    at System.Net.WebClient.GetWebResponse(WebRequest request, IAsyncResult result)
trace:    at System.Net.WebClient.DownloadBitsResponseCallback(IAsyncResult result)
trace: --- End of stack trace from previous location where exception was thrown ---
trace:    at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
trace:    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
trace:    at System.Runtime.CompilerServices.ConfiguredTaskAwaitable.ConfiguredTaskAwaiter.GetResult()
trace:    at Microsoft.Azure.Sphere.Tools.Common.Download.RecoveryImageDownloader.<DownloadAsync>d__1.MoveNext()
trace: --- End of stack trace from previous location where exception was thrown ---
trace:    at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
trace:    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
trace:    at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1.ConfiguredTaskAwaiter.GetResult()
trace:    at Microsoft.Azure.Sphere.Tools.Commands.DeviceRecoverCommand.<RetrieveImagesAsync>d__70.MoveNext()
verbose: Command completed in 00:00:02.4889126.
```

Martin1509, 2022-03-24:
Letting you know that I solved my issue. I removed fidler configuration from machine.config under

C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Config and/or
C:\Windows\Microsoft.NET\Framework\v4.0.30319\Config

**Xml part of fiddler configuration to remove:**
```
<system.net>  
    <defaultProxy>
        <proxy autoDetect="false" bypassonlocal="false" proxyaddress="http://127.0.0.1:8888" usesystemdefault="false" />
    </defaultProxy>
</system.net>
```

olegkhar, 2022-03-24:
@Martin1509 
Wow, that's so interesting.  Thanks for sharing the fix with us!

bawilless, 2022-03-24:
@olegkhar,

Is there a place that this issue and resolution can be captured for developers that may see this issue in the future?  Maybe on  Microsoft Q & A?

tokenstar, 2022-03-24:
@Martin1509  

In parallel with your investigation above, we also investigated internally and found that the errors you saw are consistent iwth not being logged in in azsphere -- i.e. the fix is to use "azsphere login".  We ahve a bug internally on fixing the error messaging here to indicate that you should log in.

However, that fix is not the same as the fix you identified above.

Can you comment?  Did you perhaps happen to use "azsphere login" around the same time as you did the fiddler changes above? thanks!

Martin1509, 2022-03-24:
@tokenstar 

I am almost sure that I didn't use "azsphere login" in deprecated CLI when I did the fiddler changes.

But I reproduced it right now, and manage extensions stops working again (didn't show online extensions) as well as Azure Sphere Explorer extension. When I tried to use "azsphere login" in deprecated CLI, I got this error:
```
error: The Azure Sphere Security Service is not responding. Please check your network connection and try again.
```

So that's obviously not the fix for my problem.

closed 2022-04-20 by Martin1509