Azure/azure-sphere-samples #234: Possible Memory Leak in get_device_id.c example

Omakas, 2022-04-11:
When running the example from get_device_id.c, I see the heap from WolfSSL not go to 0 as expected. (Due to the `deviceCert` not being freed upon function competition)

Below is what I see for WolfSSL heap usage after the function returns when monitoring heap usage. 
```
Name                         Usage (bytes)
 ==========================================
 Heap: libwolfssl             3513
 ---------------------------- -------------
```

I believe the following code needs to be added in the function to release the memory for the cert.  (Doing this call I saw the heap usage from WolfSSL go to 0 after the function call) 

```
    if (deviceCert != NULL)
    {
        wolfSSL_X509_free(deviceCert);
    }
```

Not sure if it would be best in the cleanup or somewhere else. (If added to the cleanup tag, than more code changes are required) 
But I think maybe it would be best to add it to the cleanup since the checks force the code to go to cleanup upon error.

gtrevi, 2022-04-20:
@Omakas, thanks for bringing this to our attention. Your feedback has been shared internally for further review.

gtrevi, 2022-04-20:
Closing - addressed in PR #235.

closed 2022-04-20 by gtrevi