Azure/azure-sphere-samples #238: Default Interface

BaillyAlex, 2022-07-12:
Hello,
the build is working as expect, the only issue I have is related to the network interfaces.
I setup and enabled my Wifi interfaces but instead using this one it use the network interface "azspheresvc".
Any solution :) ?

Thank you.

chandran-Az, 2022-07-12:
Hi @BaillyAlex - Are you using a specific sample for the n/w interface testing? Could you share additional information on how the wifi ids and security details are being configured? Thanks!

BaillyAlex, 2022-07-12:
Hi @chandran-Az, thank for your reply.
I'm currently using the AzureIOT sample.
About the wifi configuration I follow this step : https://docs.microsoft.com/en-us/azure-sphere/install/configure-wifi?pivots=vs-code

BaillyAlex, 2022-07-12:
The configuration of my Wifi is currently on **Configuration State enabled** and **ConnectionState Connected**

chandran-Az, 2022-07-12:
Thanks for the quick update. From a windows powershell, can you run the below command and confirm the connectivity? Ideally, it would show as connected to one of the network connections. below output is for reference.

Command - **azsphere device network list-interfaces**

Output:

[image]

BaillyAlex, 2022-07-12:
here the screenshot.
[image]

BaillyAlex, 2022-07-12:
and my issue.. :)
[image]

chandran-Az, 2022-07-12:
Thanks for sharing the screenshots. From the first screenshot, your wifi is configured and connected to Internet with IP 192.168.1.59.
When you sideload an application and attach a debugger, the connection to the device is established through FTDI port and uses the 192.168.35.1 address for the connection. This is a non-issue and just how the product works. Hope this helps.

BaillyAlex, 2022-07-12:
Do you know how could I use instead wlan0 :) ?

BaillyAlex, 2022-07-12:
Fixed with : https://docs.microsoft.com/en-us/azure-sphere/app-development/setup-iot-hub?tabs=cliv2beta#step-4-create-an-x509-device-in-your-iot-hub-for-your-azure-sphere-device

closed 2022-07-12 by BaillyAlex