Azure/azure-sphere-samples #244: Get the OS Version of the Device using Azure Sphere Device REST API

matsujirushi, 2022-07-31:
Hi,
[The Microsoft Azure Sphere Device REST APIs](https://github.com/Azure/azure-sphere-samples/blob/main/Manufacturing/src/README.md) are great!

I want to get the OS version of the device.
But I couldn't find an API.

Do you have any plans to add features?

jamesadevine, 2022-08-02:
Hi @matsujirushi,

Thanks for using the REST interface. Releasing a REST API to retrieve the OS version is currently planned. I will update the thread when it is released.

jamesadevine, 2022-08-11:
As a stopgap, @AlexanderStocks has added a gallery sample to retrieve the OS version of all connected Azure Sphere-based devices. See: https://github.com/Azure/azure-sphere-gallery/tree/main/GetDeviceOsVersion 

As I said previously, a local API will be provided in the future that does not require downloading the OS version list from Azure.

matsujirushi, 2022-08-12:
It worked. Nice! Thank you.

This sample code is enough.
Would you like to close this issue? Or keep open?

jamesadevine, 2022-08-12:
Perfect! I'll go ahead and close the issue.

Keep an eye out for the new api in an upcoming release :smile:

closed 2022-08-12 by jamesadevine

jamesadevine, 2022-09-26:
@matsujirushi with the latest OS, you can now retrieve the OS version using the dotnet or python packages directly!