Azure/azure-sphere-samples #254: VSCode: Wi-Fi details not showing up

jelalanne, 2022-09-27:
Hello,

Not sure if it's the right place to publish that bug, but didn't find a better suit.
When running the Azure Sphere Explorer for the first time, with a connected device and after claiming it with the tenant, I am trying to connect it over the network, but when listing the available networks, only the infos names' appears, like on the next capture.
Sorry it is in French, but i think the terms are quite obvious.

[image]

I suppose this is a bug. Hope you might do something about it.

jamesadevine, 2022-09-27:
Hi @jelalanne! I suppose this is as good a place than any :smile:

I assume the bug is that the extension is listing "SSID" multiple times and each contains the same network SSIDs?

jelalanne, 2022-09-27:
Hi @jamesadevine, the thing is, they are not listed at all ^^' A lot of infos are missing, and for the wi-fi subject, only key words are shown, not the actual values

jamesadevine, 2022-09-27:
Great, thanks for the additional details @jelalanne! A bug has been raised internally.

olegkhar, 2022-09-27:
Hi @jelalanne ,

Can you please turn on logging of CLI commands in the Output Window of Visual Studio?  To do so, set the environment variable "AzureSphere.PrintCliCommands" to "1"; then restart Visual Studio and re-open the Azure Sphere Explorer.

There's probably going to be a lot of output, but you can clear the output window, and then reload the Available Networks node in the Explorer, and you'll see the output of the command `azsphere device wifi scan -o json`.

That may help us understand the problem better.  Thank you!

jelalanne, 2022-09-28:
Hi @olegkhar,
I'm actually using vscode. Is there any other equivalent of your suggestions?

jelalanne, 2022-09-28:
This is what I got from Azure Sphere - UI output when refreshing details of device, does this help?

[image]

olegkhar, 2022-09-28:
Thank you, @jelalanne , this helps.

Can you please tell me if any of the network names have non-ASCII characters in them?

jelalanne, 2022-09-29:
@olegkhar  no, they are all either letters, numbers, dashes, underscores and spaces

olegkhar, 2022-10-03:
@jelalanne 
We identified the issue with the translation of our extensions and are working on correcting the translation.  In the meanwhile, can you please try changing the language to English to see if the issue still reproduces?

Thank you!

xisui-MSFT, 2022-10-06:
> @olegkhar my VScode is currently already in english. I don't know why the extension is showing up french. And I don't know how to change extension's language specific setting

There is no extension specific language setting. Displayed language is controlled by VS Code.

And I can't repro this behavior here. So my suggestion is to try to correct this with some usual approaches, for example:
- Switch to French and then switch back
- Run VS Code command "Clear display language preference"
- Restart computer
- Reinstall VS Code

If none of these fix this problem, you'll need to open an issue for VS Code: https://github.com/microsoft/vscode

jelalanne, 2022-10-07:
Hello @xisui-MSFT , switch back and forth to french worked :) Thanks!
I suppose this issue shall stay opened for you to repair the translation bug, or do I close it?

olegkhar, 2022-10-10:
Hi @jelalanne 
Thank you for confirming, and I'm happy to hear that you got the Extension working.

We'd like to keep this issue for a bit while we're fixing the Translation bug -- and we'll reply once we publish the Extension with the fix.