Azure/azure-sphere-samples #255: Reference Code Example for Azure IoT/Tools

KZeronimo, 2022-10-13:
@jamesadevine would it be possible to see how the IoTC hub endpoints are surfaced

jamesadevine, 2022-10-14:
Hey @KZeronimo!

What specifically are you interested in consuming/having visibility on from IoTHub?

KZeronimo, 2022-10-14:
Hi @jamesadevine  I would like to see the internals of the tool if possible

We have automation that creates IoTC applications, and we would like to output the DPS Scope Id and 10 IoTH endpoints that comprise the IoTC application so that our Azure Sphere device enablement team can whitelist appropriately.

I suppose we could attempt to fire the `ShowIoTCentralConfig.exe` from our automation pipeline

The key question I have at the moment is there a way to reveal the DPS Scope Id given the IoTC application URL as the only input - I notice `ShowIoTCentralConfig.exe` takes Scope Id as an input I would like that to be an output if possible.

KZeronimo, 2022-10-19:
Hi, @jamesadevine  I was playing around a bit with `ShowIoTCentralConfig.exe` - the tool is built interactive which is great; however, it would be nice to have a non-interactive mode to support automation use cases.

My goal is to be able to acquire the DPS Scope Id and 10 IoTH endpoints associated with an IoTC application in an automation pipeline and reduce the manual steps required to get the information needed to add to `app_manifest.json`

I would need to have a non-interactive mode or see how the tool retrieves the IoTH endpoints associated with the IoTC application.

jamesadevine, 2022-10-20:
Hi @KZeronimo!

Sorry for the delay.

> Hi, @jamesadevine I was playing around a bit with ShowIoTCentralConfig.exe - the tool is built interactive which is great; however, it would be nice to have a non-interactive mode to support automation use cases.

Absolutely agree, we will take a look at this.

> My goal is to be able to acquire the DPS Scope Id and 10 IoTH endpoints associated with an IoTC application in an automation pipeline and reduce the manual steps required to get the information needed to add to app_manifest.json

Completely understand. Out of curiosity, what pipeline technology are you using? GitHub Actions, ADO, other?

> One viable option is to build an input file "iotcapp.txt"

Yup, we can look at this. Is there any reason why you prefer an input text file rather than parameters to the exe?

The one drawback is that it is currently not possible to obtain the scope ID from IoTC programmatically.