Azure/azure-sphere-samples #256: Azure sphere Uart baud rate change

nishithvpoojary, 2022-11-29:
Hi. I have set the Uart baud rate in the azure starter kit as 460800. How can I change it to 115200. Does the UART library have function to change baud rate in the code. 
Thank you.

jamesadevine, 2022-11-29:
For the HL core, you should call [`UART_Open`](https://learn.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-uart/function-uart-open) with a [`UART_Config`](https://learn.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-uart/struct-uart-config) configuration struct. The configuration struct should be initialized using `UART_InitConfig`. After initialization, you can then set the `baudRate` member of the configuration struct to your desired baud rate. 

See an example here: https://github.com/Azure/azure-sphere-samples/blob/42d2d155400c92eb825d4a1e3d9c629ea2906186/Samples/DeviceToCloud/ExternalMcuLowPower/AzureSphere_HighLevelApp/uart_transport.c#L126

nishithvpoojary, 2022-11-29:
Thank you for your reply @jamesadevine .
I have already set the UART config and opened the UART.
Now I want to change the baud rate. Is it possible to change it later?
Thank you.

jamesadevine, 2022-11-30:
If you are experienced with low-level Linux calls, you should be able to modify the baud rate using [termios](https://man7.org/linux/man-pages/man3/termios.3.html). [Here is an example](https://en.wikibooks.org/wiki/Serial_Programming/termios)

An alternative is that you should `close` the file descriptor and call `UART_Open` with your revised config!

nishithpoojary, 2022-12-13:
Hi. Thank you @jamesadevine and sorry for late reply. How can I close it. Only UART open is available in UART configuration settings.

jamesadevine, 2022-12-13:
@nishithpoojary just call `close(fd);` on the file descriptor returned by `UART_Open`!

jamesadevine, 2023-05-17:
I'll do the honours then 😄

closed 2023-05-17 by jamesadevine