Azure/azure-sphere-samples #260: Exit EventLoop after DFU

nishithpoojary, 2023-05-16:
Hi. I Have used the azure sphere external MCU update sample and it is working. But I have removed the button event handler and made it direct firmware update. I want to restart the azure device after the nrf firmware update. How can I come out of event loop to do it? Thank you.

jamesadevine, 2023-05-17:
One option you have is to modify:

https://github.com/Azure/azure-sphere-samples/blob/main/Samples/ExternalMcuUpdate/AzureSphere_HighLevelApp/main.c#LL108C6-L108C27

To call [`PowerManagement_ForceSystemReboot`](https://learn.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-power/function-power-forcesystemreboot) on success.

Alternately, you could add `ExitCode_DfuSuccess` to the enum: https://github.com/Azure/azure-sphere-samples/blob/main/Samples/ExternalMcuUpdate/AzureSphere_HighLevelApp/main.c#L40

and add an if statement to check the value of exit_code here: https://github.com/Azure/azure-sphere-samples/blob/main/Samples/ExternalMcuUpdate/AzureSphere_HighLevelApp/main.c#L262

nishithpoojary, 2023-05-19:
Thank you @jamesadevine .
Calling System reboot on succes didn't work for me.
The second method of creating enum and if statement worked. Thanks a lot.

closed 2023-05-19 by nishithpoojary