Azure/azure-sphere-samples #262: 3rd party library works on Azure uart sample but doesnt work on Azure Iot sample

nishithpoojary, 2023-05-29:
I have been testing with using a 3rd party library in Azure UART sample which was working. But when I tried the same library with the Azure Iot sample, I faced several issues. I can't disclose more about this here. Is there any way to share you more details about this issue privately, may be through mail. 
Kindly let me know what I can do. Thank you.

jamesadevine, 2023-05-30:
Hi again!

You can send a support request here: azsppgsup@microsoft.com 

Otherwise, please do open issues on github hub - no judgement here.

nishithpoojary, 2023-05-30:
Hi @jamesadevine . Thanks for your reply.  Sorry that I didn't know how to get support from microsoft team privately. So had to create a issue to get support as quickly as possible. 

I used the 3rd party library in Uart sample by pasting the .c and .h file inside UART folder and putting the .c file name in "add_executable(${PROJECT_NAME}" of the CMakeLists.txt file. This worked.

When I used the same library in Azure IOT sample by pasting both .h and .c library files inside "common" folder and typing both .c and .h in "${CMAKE_CURRENT_LIST_DIR}/" of the CmakeLists.txt in common folder,  I get errors in .c library. 
I couldn't understand what caused the issue.

nishithpoojary, 2023-05-30:
I am using the Azure IoT sample now. I am trying to use the 3rd party library in this project. I have already added the details of library in cmakelists present inside the common folder and pasted the .c and .h files in the common folder.
subdirectory is already available for common folder.