Azure/azure-sphere-samples #264: Is it possible to clear Uart in azure

nishithpoojary, 2023-06-30:
Hi. I have to send some hex values through uart.  But I noticed even after the data is sent , when I send next set of hex values it had some unwanted values. 
example :- data to be sent 02 04 01 00
data received by FTDI connected to azure : E0 02 04 01 00
The dont understand how a extra byte E0 (this value changes) is received by FTDI.

Please guide me on this issue. Thank you.

nishithpoojary, 2023-06-30:
While the application is running itself I faced this issue. Even after erasing the azure board and tried deploying I faced this issue.

jamesadevine, 2023-06-30:
I doubt the board is actually sending this data intentionally. Try flushing the FTDI buffers before you try and read from the device.

nishithpoojary, 2023-06-30:
Its the same when i use nrf board connected to azure. A specific hex value is required by nrf to do it work. But since the nrf doesnt receive this proper hex value, it work start its functions.