Azure/azure-sphere-samples #268: Wifi Ble Sample Pair key Issue

nishithvpoojary, 2023-11-03:
Hi. We use the sample https://github.com/Azure/azure-sphere-samples/tree/main/Samples/WifiSetupAndDeviceControlViaBle for BLE connection. After connecting the Azure BLE from windows app, we dont see any "enter the pass key" tab. Is this a windows app issue or device to windows app communication issue? 
I tried to debug the "ble_app_uart_pca10040_s132" SES file, but i got the below build error.

**Building ‘ble_app_uart_pca10040_s132’ from solution ‘ble_app_uart_pca10040_s132’ in configuration ‘Release’
2> Assembling ‘thumb_crt0.s’
3> Checking Source File Exists
3> C:/azure-sphere-samples-main/Samples/WifiSetupAndDeviceControlViaBle/Nrf52App/pca10040/s132/ses/CHANGE_THIS_TO_YOUR_NORDIC_SDK_PATH/components/libraries/log/src/nrf_log_backend_rtt.c does not exist.
Build failed**

Kindly help me with this issue. Thank you.

jamesadevine, 2023-11-20:
Did you read step 1ii from here? https://github.com/Azure/azure-sphere-samples/blob/96fc529d76e3a964b0ad75d5abebdad8acaf7695/Samples/ExternalMcuUpdate/README.md#build-your-own-bootloader

nishithvpoojary, 2023-11-21:
Sorry... my mistake. after changing the path the still i faced issue with the nrf sdk version 17. 
So i used the version 15. I can build the code now. Thank you.
 I will be back here after testing the app pairing. Thank you