Azure/azure-sphere-samples #269: Wifi disabled automatically in azure sphere

nishithvpoojary, 2023-11-22:
Hi. In the Sample [https://github.com/Azure/azure-sphere-samples/tree/main/Samples/WiFi/WiFi_HighLevelApp](url) after adding the wifi and enabling it, if i disconnect the device without doing the remaining functions( i.e., the disable , duplicate and delete ) and connect the device back again.., why does the wifi state shows disabled.
The wifi state stays enabled if i add it through command prompt, but if it is done through code, it becomes disabled automatically. How is it? Thank you.

jamesadevine, 2023-11-22:
Are you calling [WifiConfig_PersistConfig](https://learn.microsoft.com/en-us/azure-sphere/reference/applibs-reference/applibs-wificonfig/function-wificonfig-persistconfig?view=azure-sphere-legacy) in your code?

nishithvpoojary, 2023-11-22:
Yes. Is that the function causing this ? Thanks for your reply @jamesadevine

jamesadevine, 2023-11-22:
Well, I think minimally you need to:

1) Call WifiConfig_AddNetwork 
2) Call the WifiConfig_Set* calls for the network ID returned in 1).
3) Call WifiConfig_SetNetworkEnabled 
4) Call WifiConfig_PersistConfig

Are making all the calls listed above?

nishithvpoojary, 2023-11-22:
i checked it now.... WifiConfig_PersistConfig was missing inside the WifiNetworkEnableState(void).. Now it is working perfectly. Thanks a lot.

jamesadevine, 2023-11-22:
Perfect! no problem

closed 2023-11-22 by jamesadevine